ARM-software/CMSIS_5 #428: [Enhancement] [Device] Cortex M33 with DSP and no FPU unit is not considered

fgr1986, 2018-09-04:
The current structure of Device folder files for Cortex M33 does not consider devices with DSP but no FPU, and no TZ.

CM33 considered options:
1. No FPU no DSP no TZ
2. No FPU no DSP with TZ
3. FPU and DSP and TZ
4. FPU and DSP no TZ

The new device type would have the following flags in the (new) `ARMCM33_DSP.h` file:
```

#define __CM33_REV                0x0000U   /* Core revision r0p1 */
#define __SAUREGION_PRESENT       0U        /* SAU regions present */
#define __MPU_PRESENT             1U        /* MPU present */
#define __VTOR_PRESENT            1U        /* VTOR present */
#define __NVIC_PRIO_BITS          3U        /* Number of Bits used for Priority Levels */
#define __Vendor_SysTickConfig    0U        /* Set to 1 if different SysTick Config is used */
#define __FPU_PRESENT             0U        /* FPU present */
#define __DSP_PRESENT             1U        /* DSP extension present */
```
Please note that the file, together with the possible modification in ` system_ARMCM33.c`  shold also be compatible with DSP lib (see issue https://github.com/ARM-software/CMSIS_5/issues/427)

Possible additions in ` system_ARMCM33.c`
```
...
#elif defined (ARMCM33_DSP)
  #include "ARMCM33_DSP.h"
#else
  #error device not specified!
#endif
```

JonatanAntoni, 2018-09-05:
Hi @fgr1986,

Thanks for your feedback. The "pseudo" devices you refer to are only intended to be used as some reference for device vendors. Hence we considered only the configurations assumed to be widely adopted.

Do you have a concrete scenario in mind you would need yet another pseudo device header for? From pure software/compilation point of view you should get the correct noFPU+DSP result by simply not using the FPU in your application, i.e. by setting `-mfloat-abi=soft [-mfpu=none]`.

Cheers,
Jonatan

fgr1986, 2018-09-05:
Thank you @JonatanAntoni , as you suggest we can use `ARMCM33_DSP_FP.h` and compile with the following collected gcc flags:
`COLLECT_GCC_OPTIONS='-mthumb' '-march=armv8-m.main+dsp' '-mfloat-abi=soft' '-D' 'ARMCM33_DSP_FP' '-D' 'ARM_MATH_CM33' '-D' '_DSP' '-D' '__DSP_PRESENT=1' '-fno-inline-functions'`

closed 2018-09-05 by fgr1986