ARM-software/CMSIS_5 #434: Negative timer tick

juliobc, 2018-09-21:
Hi, I saw an unusual behavior of a os timer and I have this in the RTOS window when I am debuging:

[image]

It is possible to have negative numbers?
Thanks

juliobc, 2018-09-21:
I am sorry, I am working with the last version of RTOS RTX5

JonatanAntoni, 2018-09-21:
Hi @juliobc,

That looks like an error in Component Viewer on the first glance. Both values, `osRtxTimer_t.tick` and `osRtxTimer_t.load` are defined unsigned. I need to check whether the display accidentally uses signed types instead. In this case large unsigned values would appear as negatives.

Thank you very much for making us aware of that issue. I'll report back soon.

Cheers,
Jonatan

JonatanAntoni, 2018-09-21:
Addendum: I quickly glanced over RTX5.scvd (which defines the contents for the component viewer window). The display string for `Tick` and `Load` are both using `%d` format specifier, which is signed integer. As a workaround you could try to change this locally using `%u` specifier instead.

JonatanAntoni, 2018-09-21:
HI @juliobc,

Unfortunately I got notified by the developers that `%u` is not accepted by MDK. Sorry for that. We need to further investigate where we need to apply the fix.

Cheers,
Jonatan

juliobc, 2018-09-21:
Hi @JonatanAntoni 
ok. I revert
Thanks

JonatanAntoni, 2019-01-09:
Hi @juliobc,

we have now traced the issue back into the tool. Hence its not a bug here.
Unfortunately, you need to wait for next release of MDK which will be 5.27.

Cheers,
Jonatan

juliobc, 2019-01-10:
Ok @JonatanAntoni
thanks.

JonatanAntoni, 2019-05-21:
Hi @juliobc,

May I ask you to finally check the issue is solved with MDK 5.27, please?
Consider closing this issue if no further discussion required.

Thanks,
Jonatan

juliobc, 2019-05-21:
Fixed in MDK 5.27

Thanks

closed 2019-05-21 by juliobc