ARM-software/CMSIS_5 #435: How can I set the OS_IDLE_THREAD_NAME?

juliobc, 2018-09-26:
Hi, I see that exist the code:

`#if defined(OS_IDLE_THREAD_NAME)
  OS_IDLE_THREAD_NAME,
#else
  NULL,
#endif`

then, how can I set the OS_IDLE_THREAD_NAME? I want to identify that thread when I execute the "osThreadEnumerate()" and the "osThreadGetName(thread_array[i])"

Thanks

juliobc, 2018-09-26:
and the same with OS_TIMER_THREAD_NAME

JonatanAntoni, 2018-09-26:
Hi @juliobc,

The easiest way to identify the threads is to use their numeric thread ids, see `rtx_os.h` in `osRtxInfo`.
If you want to set the names you can simply set those defines, e.g. using the `RTX_Config.h`.

Does this solve your question?

Cheers,
Jonatan

juliobc, 2018-09-26:
Hi @JonatanAntoni 

Where in the RTX_Config.h can you set the OS_IDLE_THREAD_NAME and OS_TIMER_THREAD_NAME? I can not in $Revision:   V5.4.0

How do you know the id of TIMER_THREAD ?

thanks

juliobc, 2018-09-26:
I can set the defines OS_IDLE_THREAD_NAME and OS_TIMER_THREAD_NAME. Ok. The wizard does not show it and when I update the RTX version this defines disappears

How do you know the id of TIMER_THREAD ?