ARM-software/CMSIS_5 #437: Order of input tensor to convolutional layers

ColinNordin, 2018-09-26:
I can't find any information about in which order the input tensor to convolutional layers should be.

If I have a an input tensor of shape (row, col, ch) = (20, 40, 1), should the input buffer then be [row1, row2, ..., row20] or should it be [col1, col2, ..., col40]? And if I have more than one channel the possibilities increase.

SiarheiFedartsou, 2018-11-21:
@ColinNordin I think you can use this line as a reference https://github.com/ARM-software/ML-KWS-for-MCU/blob/e3f1685b406c0cc2c5724d2278e8bf720b009cce/quant_test.py#L110
You know the initial order in TensorFlow (Width -> Height -> Input Channels -> Output Channels) and after reordering it should be (Output Channels ->  Width -> Height -> Input Channels).

felix-johnny, 2021-01-07:
@muskedunder  Some late tidying up of tickets. 

I guess you asked for the legacy(https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN#legacy-vs-tfl-micro-compliant-apis)  API's. Since that is not the supported one, I'll answer it for the TFLu based API's for future reference. Information about the data format for input, filter and output tensor is provided along with the function prototype like for e.g https://github.com/ARM-software/CMSIS_5/blob/9ce90ae0d94ca6893dc6e03c53d8569e17eecedc/CMSIS/NN/Include/arm_nnfunctions.h#L207 . 

With that, I'll close this ticket.

closed 2021-01-07 by felix-johnny