ARM-software/CMSIS_5 #438: setting timer callback queue

juliobc, 2018-09-27:
Hi,

I actually have an osRtxErrorTimerQueueOverflow.

I would like to know if it is convenient to configure that queue with larger values than necessary or if there are problems with large values.

Thanks

JonatanAntoni, 2018-09-27:
Hi @juliobc,

This error denotes that your timers create more callbacks than the application is able to handle. If this happens occasionally it might only be a temporary jitter in callback processing. Such a case could be covered by increasing the timer callback queue size. Please keep in mind that a larger queue leads to larger time delays between the actual timer expiry and the effective application notification.

If you observe queue overflows periodically you might have an overload issue, i.e. too much callbacks. Please take into account that all you timer callbacks are executed in the timer thread context. All callbacks are processed in order of timer expiry, one by one. Hence it's highly recommended to consider the same restrictions for the callback routines as for interrupt routines, i.e. you should not use blocking calls (even if they are possible). Instead delegate lengthy or blocking activities to another (worker) thread. 

Does this answer your question?

Cheers,
Jonatan

juliobc, 2018-09-27:
ok,
Thank you very much

juliobc, 2018-10-01:
Hi @JonatanAntoni 

I still having osRtxErrorTimerQueueOverflow with Timer Callback Queue entries = 15 (higher than number of timers). The Timer that throws this error (object_id in osRtxErrorNotify()) only set an event flag inside its callback. 

This error is rarely released but I need to solve this problem. Any ideas?

Thanks

JonatanAntoni, 2018-10-01:
Hi @juliobc,

The timer that causes the overflow might not be the cause for filling the queue. Please inspect

- the timer queue, i.e. which callbacks are still pending.
- what state the timer thread is in.

You might recognize that another timer is running too fast. Or that one of your timer callbacks blocks the timer thread for too long. Or even that the priority configuration is incorrect, i.e. the timer thread does not get the precedence by using the highest priority. 

Without detailed knowledge of your application its hard to tell you more. Is it feasible for you to provide a reproducer example?

Cheers,
Jonatan

juliobc, 2018-10-01:
Hi @JonatanAntoni 

It is almost impossible to create a reproducer example. This error occurs once every three days on average.

I have a thread with critical priority. This thread performs an ADC capture with a very high sampling during 200 msec. This error always occurs during this ADC capture, but I do not understand why this priority thread could interfere with the timer thread.

juliobc, 2018-10-08:
Hi @JonatanAntoni 

This weekend I captured the event and now I have the fw stopped and debugging in the osRtxErrorNotify() with code error = osRtxErrorTimerQueueOverflow

I attach some captures:

First, in the RTX RTOS window I see that ISR FIFO queue is full !!!

[image]

and the Idle Thread is running and I can not see its used stack:

[image]

the Timer Thread is blocked. And the info of timers is this:
[image]

Finally, in the osRtxInfo can see the timer info:
[image]

I suspect that the problem could be the ISR FIFO queue, not the TimerQueueOverflow

I have the fw stopped and debugging. Any idea?

Thanks

juliobc, 2018-10-08:
I am sorry, I closed accidentally

JonatanAntoni, 2018-10-08:
Hi @juliobc,

That becomes a totally different story, now.

The ISR queue is used to defer scheduling from user interrupts to PendSV. An overflow to this queue indicates a problem with interrupt load. It looks like PendSV does not get enough processing time to handle all the queued scheduling jobs. 

Each slot in the ISR queue is the pointer to an RTOS-object (like thread or message queue) that needs post-processing after modification from user interrupt. It looks like there are a lot of different objects receiving triggers from interrupts. If you think this is expected behavior you might want to enlarge the ISR queue.

Cheers,
Jonatan

juliobc, 2018-10-08:
Hi @JonatanAntoni 
Then the info of TimerQueueOverflow is wrong? I think this is not the expected behavior. The tread with critical priority is launching an interrupt that only copy the captured data to the memory. How can I know which RTOS-object is modified? Is possible to save all the actual RTOS info to an txt?

Thanks

JonatanAntoni, 2018-10-08:
HI @juliobc,

I admit sorting out the RTOS objects from the ISR queue is a bit cumbersome. You need to compare the object ids manually. In fact RTX5 simply uses the pointers to the control blocks as object ids. Hence you should be able to find the modified object by comparing the values in the ISR list with all your RTOS objects, i.e. threads. 

Another staring point for your analysis can be all your user interrupt functions. Which of them make use of RTOS calls, like message queue put or event set? 

I am not sure where the original TimerQueueOverflow information came from. You might have different issues in your application. Without more detailed knowledge about your application its hard to tell. The only thing I can suggest is to analyze your application piece-wise. Perhaps tools like EventRecorder, SystemAnalyzer and Exception tracing might help as well. Please get in touch with the customer support if you need additional help on these commercial tools.

Cheers,
Jonatan

juliobc, 2018-10-09:
Hi @JonatanAntoni 

I saw every RTOS object and I can not find any object with the address equal to the address of the ISR FIFO slot. Could be a pointer overflow?
Also, I configured that ISR FIFO queue to 48 and I encontered the same osErrorTimerQueueOverflow. There are a few timers running and they have a lot of seconds to expire.
In my interrupts I only set an event flag too.

This error is very difficult to find because I do not an expert inside the RTOS. I can suspect that the implementation of the low power in the Idle Thread could be guilty because when I debugged this error (in the images) that Thread was running. Could you send me any cue? The low power implementation in the Idle Thread starts with osKernelSuspend() and end with osKernelResume(tc_wakeup). I am testing a fw without low power implementation to verify if this is the cause of the error (it is running for 16 hours).

Thanks

JonatanAntoni, 2018-10-09:
Hi @juliobc,

Phew. I have now discussed the issue with the RTOS experts in more detail.
We finally came up with the conclusion that the ISR queue overflow seems to be a consequence of being stuck in osErrorNotify which blocks the SysTick interrupt from completing. We are majorly concerned about the timer configuration we can see from your screen shots.

It looks like you have >8 timers in your system. The timer queue is configured to 8 slots. Hence if more than 8 timers expires at once you'll instantly get the timer queue overflow. Please try to enlarge your timer queue to a value larger than the number of timers you have in the system. Assure that the ISR queue is even larger because each expiring timer puts a message to the timer queue which also gets put to the ISR queue for post-processing. 

In order to analyze your issues further please try to use EventRecorder. Configure EventRecorder in a way that we can observe the calls to timers and message queues. If you end up in osErrorNotify again please export the EventRecorder log to CSV and provide it to us. That would help us to understand whats going on in your system much better.

Cheers,
Jonatan

juliobc, 2018-10-09:
Hi @JonatanAntoni 

How can you observe that thre are 8 slots in Timer Queue? I have configured 15 in RTX_Config.h

[image]

I have more slots than timers. Also, at the moment of the error, only 2 timers are running, the others are stopped (to see images). I will configur the eventRecorder and I will try capture the error again

Thank you very much

JonatanAntoni, 2018-10-11:
Hi @juliobc,

If you state you have configured 15 slots we might have a misunderstanding here, sorry. 
Unfortunately your screenshots above did not reveal the actual size of the memory pool used for timer messages. The only explanation we came up with is that allocating another message slot in the queues memory pool fails. 

You could try to reveal the actual cause for `osErrorTimerQueueOverflow` by taking a closer look to the call stack. Which internal condition caused the call to `osErrorNotify` using this code?

The EventRecorder log could help to get a better understanding of what happened before.

Cheers,
Jonatan

juliobc, 2018-10-15:
Hi @JonatanAntoni 

I captured the breakpoint again and I attaach the EventRecorder. 
[Event_Recorder.zip](https://github.com/ARM-software/CMSIS_5/files/2478145/Event_Recorder.zip)

The Idle thread is running when I capture the breakpoint.
The Idle Thread implements a low power mechanism like this:

```
while(1) {
		tc_wakeup = osKernelSuspend();
		if (tc_wakeup > 0) {
			/* ... */
			if (tc_wakeup > MAX_TIME_WITH_VERY_LOW_POWER_FORBIDDEN_IN_MSEC)
				HAL_PWR_EnterSTOPMode(PWR_LOWPOWERREGULATOR_ON, PWR_STOPENTRY_WFE);
			else
				HAL_PWR_EnterSLEEPMode(PWR_LOWPOWERREGULATOR_ON, PWR_SLEEPENTRY_WFE);
			
			/* ... */
			HAL_SuspendTick();
		}
		osKernelResume(tc_wakeup);
	}
```
The call stack of Idle Thread is:

[image]

Can you give me a clue?
Thanks

JonatanAntoni, 2018-10-15:
Hi @juliobc,

It is still not obvious to me whats going wrong. Your EventRecorder log lacks the information about your timers. But there is another aspect that makes me wondering.

Can you inspect `osRtxInfo.timer.list`, please? This pointer denotes the head of the linked list of active timers. If you have only one timer active it should contain only a single element. Your call stack makes me think this list contains more timers or a cyclic loop.

The cause for the `osRtxErrorTimerQueueOverflow` is in `svcRtxKernelResume`. This portion of the code assures that all software timers are correctly decremented and signaled after returning from low power. This is done by walking through the linked list until either
* all ticks have been replayed, or
* the end of the list is reached.

In your case it seems none of the above criteria is hit. Instead an unbound amount of timers is being signaled. This inevitably leads to a queue overflow.

Cheers,
Jonatan

juliobc, 2018-10-15:
Hi @JonatanAntoni 

this is the list:
[image]

and those are the timers I have:

[image]

only 3 timers are runnig: 0x20005428 (just with tick=0), 0x20014348 and 0x20014410. How do you know how many timers are in the list?

Thanks

JonatanAntoni, 2018-10-15:
Hi @juliobc,

It's a double-linked list, i.e. each item has `next` and `prev` pointers. Please follow the chain of `next` pointers until you find the terminating `NULL` one. Do you observe to many members in that list? Or any cyclic loop, i.e. never running into a `NULL` pointer down the chain?

At last there must be a reason why the memory pool used internally by the timer message queue is depleted, i.e. finally causing the `osRtxErrorTimerQueueOverflow` notification.

Cheers,
Jonatan

JonatanAntoni, 2018-10-15:
Hi @juliobc,

I just recognized another reason that could lead to the overwhelming amount of signaled timers.
The timer you are using is `periodic`. Hence it is automatically restarted and added back to the linked list after it has been signaled.

If your actual sleep time covers multiple timer periods this timer would be signaled once per period spent in low power mode. Depending on your actual values this could overwhelm the queue as well. Can you navigate up your call stack and watch the value of local variable `sleep_ticks` in `svcRtxKernelResume`?

Admittedly, I am poking about in a fog.

Cheers,
Jonatan

juliobc, 2018-10-15:
Hi @JonatanAntoni 

there are only 2 timers in the list:
[image]

and the timer 0x20005428 is not in the list and it has a long periodic time:
[image]

JonatanAntoni, 2018-10-15:
Hi @juliobc,

Thanks for clarifying this. The timer list looks good to me. 

If you inspect the code of `svcRtxKernelResume` and `osRtxTimerTick` you can see what happens with your periodic timer. Lets assume a very large time spent in power down mode, e.g. 3'000'000 ticks (10 periods of your timer). Calling `osKernelResume` with this value would lead to:

1. TimerUnlink
2. osMessageQueuePut
3. TimerInsert
4. Repeat steps 1-3 nine times
5. Unblock the Kernel
6. Timer thread processes 10 signals for the timer

Hence this scenario would consume 10 slots in the timer message queue. If the message queue depletes during above example you get:

1. TimerUnlink
2. osMessageQueuePut
3. osRtxErrorNotify

And the result would look exactly like what I can see in your data so far.

Can you analyse your call stack up to `svcRtxKernelResume`, please? This function has a local variable `sleep_ticks` that tells you how many ticks are left for replay.

Cheers,
Jonatan

juliobc, 2018-10-15:
Hi @JonatanAntoni 

It is not in scope
[image]

maybe it is a local variable?
Thanks

JonatanAntoni, 2018-10-15:
Yes, it is a local variable. I expect it was pushed to the stack somewhere. 
Is it possible to move the `PC` to somewhere behind the endless loop in `osRtxErrorNotify` in order to get the application to run back up to `svcRtxKernelResume`? You should be able to put a breakpoint right after `osRtxInfo.timer.tick()`. From there the variable should be in scope and visible.

juliobc, 2018-10-15:
Hi @JonatanAntoni 

unfortunately, after osRtxErrorNotify, the programm threw a HardFault exception and the stack pointer went to HardFault_Handler()

I can reset and wait for the error again.
Thanks

juliobc, 2018-10-15:
Hi @JonatanAntoni 

colud you see the sleep_tick in the Event Recorder?
Thanks

JonatanAntoni, 2018-10-15:
Yes, its logged alongside with `EvrRtxKernelResume`.

juliobc, 2018-10-15:
Hi @JonatanAntoni 
yes, all events are enabled actually
[image]
[image]
[image]

Thanks

juliobc, 2018-10-17:
Hi @JonatanAntoni 

I am trying capture a new breakpoint.

Meanwhile, do you have any cue of problem? We know that the queue is not full (only have 3 timers) and the error is when one timer is signaled (wich it has a long periodic time compared with the time in low power) while the IdleThread is running entering in low power with kernel suspend and resume.

Do you think that I could put any new Event Recorder sentence in any point of code? For example for show the ticks you wanted discover?

Thanks

JonatanAntoni, 2018-10-17:
Hi @juliobc,

Unfortunately I don't have anything new. The error code you are seeing is raised due to depletion of the timer message queues memory pool. We need to figure out how that happens.

Cheers,
Jonatan

juliobc, 2018-10-24:
Hi @JonatanAntoni 

I captured the error again:

[image]

[image]

[image]

The ISR_Queue is full. And the pointers in that queue has 20 bytes of difference from one to another. What is this? Could it be an overflow pointer?
Thanks

JonatanAntoni, 2018-10-24:
Hi @juliobc,

Can you please go further up the call stack? I suppose you are in `svcRtxKernelResume`, aren't you? What is the value of parameter `sleep_ticks` this function was called with?
Please take special note of the loop where `osRtxTimerTick` is called in. I can imagine that this loop does not run to completion.

The pointers are 20 bytes apart because each denotes a single message, all messages are placed next to each other in the same memory pool. Please inspect the memory between addresses `0x20026264` and `0x20026390`. You can use a watch window and cast all the pointers to `os_message_t*`. The field `id` should be equal to `osRtxIdMessage` (0x07). I suppose the message body (last 8 bytes, type `osRtxTimerFinfo_t`) all contains the same values (pointer to your callback function).

Cheers,
Jonatan

juliobc, 2018-10-25:
Hi @JonatanAntoni 

I have a new breakpoint in the same point. I analized themessages of the ISR_FIFO. I show you the result:
[image]

Thre is a osRtxIdEventFlags (0x03) in the middle.

Thre is not a osRtxTimerFinfo_t in the last 8 bytes. 
[image]

I see that the struct of os_message_t is:
```
typedef struct osRtxMessage_s {
  uint8_t                          id;  ///< Object Identifier
  uint8_t              reserved_state;  ///< Object State (not used)
  uint8_t                       flags;  ///< Object Flags
  uint8_t                    priority;  ///< Message Priority
  struct osRtxMessage_s         *prev;  ///< Pointer to previous Message
  struct osRtxMessage_s         *next;  ///< Pointer to next Message
} osRtxMessage_t;
```
Then I do not understand what "struct osRtxMessage_s *prev;" is. In the image I can see that *prev = 0x20005440, and this I think is not another osRtxMessage_s. 

I can see in the RTX RTOS wondow:

[image]

if this timer has the struct:

```
/// Timer Control Block
typedef struct osRtxTimer_s {
  uint8_t                          id;  ///< Object Identifier
  uint8_t                       state;  ///< Object State
  uint8_t                       flags;  ///< Object Flags
  uint8_t                        type;  ///< Timer Type (Periodic/One-shot)
  const char                    *name;  ///< Object Name
  struct osRtxTimer_s           *prev;  ///< Pointer to previous active Timer
  struct osRtxTimer_s           *next;  ///< Pointer to next active Timer
  uint32_t                       tick;  ///< Timer current Tick
  uint32_t                       load;  ///< Timer Load value
  osRtxTimerFinfo_t             finfo;  ///< Timer Function Info
} osRtxTimer_t;
```

then finfo = 0x20005440
Is this ok?

JonatanAntoni, 2018-10-25:
Hi @juliobc,

Lets ignore the `osRtxIdEventFlags` in between for the moment. I suppose you use an EventFlags object somewhere. That should be fine anyway.

The interesting part it the actual message body which is behind the message header you already see. The messages are all 20 bytes in size. The header `osRtxMessage_t` occupies the first 12 bytes. The pointers `prev` and `next` are used to link messages into a priority-sorted list when added to a queue. We can ignore that for now.

We are interested in the message payload, i.e. the 8 bytes following the header. The payload contains the `osRtxTimerFinfo_t` from the timer that needs to be signaled, i.e. the function pointer and argument value. I suppose that for all of the messages stuck in the `isr_queue` the `finfo` belongs to the same timer.

If this is the case your device was suspended for multiple timer periods. That's why I am still asking for the value of `sleep_ticks` given to `osKernelResume` call? The EventRecorder log might reveal this.

Cheers,
Jonatan

juliobc, 2018-10-25:
Hi @JonatanAntoni 

that timer has a load value of 3000

[image]

In this case I see an event recorder a bit different:

[event_recorder.zip](https://github.com/ARM-software/CMSIS_5/files/2514025/event_recorder.zip)

Can you see any cue?
Thanks

JonatanAntoni, 2018-10-25:
Hi @juliobc,

I cannot see anything relevant in your Event Recorder log.

Again: What is the value of parameter `sleep_ticks` of your current call to `osKernelResume`? The Event Recorder log lacks this information. How do you calculate the value for `sleep_ticks`? Do you always use the fixed value 5 returned from `osKernelSuspend`? 

What are the contents of the `irq_queue` messages? Do they all refer to your periodic timer as I suppose?

Cheers,
Jonatan

juliobc, 2018-10-25:
I can not see the sleep_ticks parameter. It is not in scope after osRtxTimerTick return (jumping the (void)osRtxErrorNotify(osRtxErrorTimerQueueOverflow, timer);)

Teorically, that value should be 5
Could I save the value to a static variable when the svcRtxKernelResume(uint32_t sleep_ticks) is launched.

Thank

juliobc, 2018-10-25:
Hi @JonatanAntoni 

[image]

you can observe in this image that I jumped the osError and back to the svcRtxKernelResume() and I stay now in line 455 and I can not see the sleep_tick in the call Stack + Locals window

I will save that value in a static variable and I will try capture the error again

Thank you for your support.

juliobc, 2018-10-26:
Hi @JonatanAntoni 

I captured the paremeter sleep_ticks in the function void svcRtxKernelResume (uint32_t sleep_ticks) when the error in captured. The method to cature the value is this:

[image]

When I captured the error I can see sleep_ticks_bkp = 0xFFFFFC19. But this value is wrong because in the event recorder I can see the short duration in IdleThread. The value 0xFFFFFC19 looks like a signed value.

[image]

[image]

It could be the problem. I will investigate about signed and unsigned variables in low power.

JonatanAntoni, 2018-10-26:
Gotcha 🥇 

Calling `osKernelResume(0xFFFFFC19)` would instantly trigger your periodic timer more than 1.4 million times. I guess it's obvious that this overwhelms the timer queue.

I recommend inserting some sanity checks before calling `osKernelResume` with the calculated value. If the value is much larger than your timer period (3000 ticks) something went wrong. Hence you need to decide how to recover, e.g. by limiting the value of `sleep_ticks` to some maximum.

Cheers,
Jonatan

juliobc, 2018-10-26:
Hi @JonatanAntoni 

ok, I will continue searching the error 
Thank you

juliobc, 2018-10-29:
Hi, @JonatanAntoni 
Finally I solved the error. The problem was in the ST HAL libraries when the function to obtain the RTC time returned an unsynchronized value between the time and the subseconds, so, the difference between the time of start and end into the suspended kernel function had result negative.

Thank you very much for your support. You've been very helpful

closed 2018-10-29 by juliobc