ARM-software/CMSIS_5 #442: osDelay() for 1 msec

juliobc, 2018-10-02:
Hi,

I am having a delay of 1,6 msec when I put osDelay(1); I do not understand this behavior. The API shows:

"The actual time delay may be up to one timer tick less than specified, i.e. calling osDelay(1) right before the next system tick occurs the thread is rescheduled immediately."

When I put osDelay(2); I am having 2,6 msec. 

Is this correct?

Thanks

JonatanAntoni, 2018-10-02:
Hi @juliobc,

Actually that sounds surprising to me as well. `osDelay(1)` blocks the current thread and puts a `1` to the timeout. At the next system tick the counter is reduced by `1`, reaches `0` and the thread should be reactivated. Having 1 tick = 1 msec using `osDelay(1)` should give you an actual delay of up to 1 msec.

Do you have other threads running in parallel? If you have another thread at the same or higher priority this might cause the unblocked thread not to be scheduled immediately.

You should be able to track whats actually going on using RTOS-aware debugging or EventRecorder. Or by inspecting the control blocks, manually.

Cheers,
Jonatan

juliobc, 2018-10-05:
Hi @JonatanAntoni 

The problem was into Idle Thread. The function SetWakeUpTimer() of HAL ST was delayed by 0.5 msec more than expected.

Regards

closed 2018-10-05 by juliobc