ARM-software/CMSIS_5 #444: CMSIS-NN Output verification for examples

mjs513, 2018-10-02:
I have just moved CMSIS-NN into the Teensy environment (K64/K66).  The examples run but I do have a couple of simple questions.  If this is not the right place to do this please tell where the best place would be.

Q1.  In the CIFAR10 example there are two datasets that you provide in the .h file.  Can you tell me what the output is expected for each example?
Q2.  When I run the GRU example all I get from the example is it completed iteration one and iteration two.  Some how would think you should provide an example that outputs the well output data.

v/R
Mike

JoeriRethy, 2018-12-07:
Worked with Cifar10 so i can answer that question..

It predicts which picture  you uploaded. In the input header file you can find the binary form of that image pixelized in 32x32.. With python you can convert it easily to that format en print that string out.. 

```
def resizeImage(srcfile, new_width=32, new_height=32):
    '''
    Resize and crop a image to desired resolution and return the 
    data as HWC format numpy array.
    srcfile: the source image file path.
    new_width: new desired width.
    new_height: new desired height.
    '''
    pil_image = Image.open(srcfile)
    pil_image = ImageOps.fit(pil_image, (new_width, new_height), Image.ANTIALIAS)
    pil_image_rgb = pil_image.convert('RGB')
    return np.asarray(pil_image_rgb).flatten()
```
Copy paste that string into the header file and the 0,1,2,3,4,... will be the prediction of the output... You might want to read this http://www.cs.toronto.edu/~kriz/cifar.html.

etiennedemontalivet, 2019-02-01:
Hello everyone,

I just tried the CMSIS cifar10 example on STM32H7 and I am although wondering if any results can be expected from this NN.
I have extracted the first 100 images and corresponding label from the _test_batch.bin_ (see [here](http://www.cs.toronto.edu/~kriz/cifar.html)). After putting those images into the [CMSIS cifar10 example](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN/Examples/ARM/arm_nn_examples/cifar10), I get less than 25% well classification.
My guess was to take the higher _output_data_ index as the detected class.

Q1 : Did I guess well about interpreting the output ?
Q2 : Is there any results with those parameters and weights for comparison ?

Thank you !

Etienne

llefaucheur, 2019-02-01:
Hi Etienne,
CIFAR from CMSIS-NN is under dimensionned for the problem. You need to increase the size of the network and retrain it. This problem was reported here : https://github.com/ARM-software/CMSIS_5/issues/452 
Regards, Laurent

etiennedemontalivet, 2019-02-01:
Hi Laurent,
Thank you for pointing that out.
Regards

closed 2021-01-07 by felix-johnny