ARM-software/CMSIS_5 #449: thread id's of the idle and the timer thread

joabeck, 2018-10-09:
Dears Sirs,
i want to know the free stack space of all the threads in my software (runtime not only in the debugger). For the user threads i can get the stack space with the function osThreadGetStackSpace. This function needs the thread id as parameter.
How can i get the thread id's of the idle and the timer threads?

Thank you for your support.
Best regards
Joachim Beck

JonatanAntoni, 2018-10-09:
Hi Joachim,

We currently don't have a public API to access these internals. But you can use the pointers from `osRtxInfo` (the global RTOS control block). The idle thread is linked by `osRtxInfo.thread.idle` and the timer thread by `osRtxInfo.timer.thread`.

Does this solve your requirements?

Cheers,
Jonatan

joabeck, 2018-10-15:
Hi Jonathan,
i can live with this, but i think this is not portable, if one wants to use another CMSIS RTOS2.
Thank you
Cheers
Joachim

JonatanAntoni, 2018-10-15:
Hi Joachim,

Yes, of course this is not portable. In fact having timer and idle threads is some kind of implementation specific, anyway. Hence I don't see how one could ever make this really portable.

We expect portable software components (e.g. communication stacks) relying on a limited subset of RTOS primitives not having a dependency like this.

Cheers,
Jonatan