ARM-software/CMSIS_5 #451: Is it possible to change the kernel tick after starting the kernel?

joabeck, 2018-10-15:
Is it possible to change the kernel tick (SYSTICK) after starting the kernel? This could be helpful to save power.
If yes, what is the preferred Methode to do so?

Thank you for your support.
Regards
Joachim

joabeck, 2018-10-15:
Hi Jonathan, 

Sorry, my question was wrong. I do not want to change the kernel tick, i want to reduce or raise the microprocessor core clock, which changes also the SYSTICK frequency. 
I saw in the source code, that the SYSTICK is setup only in osKernelStart. My question is now, what has to be done, if i want to change the core clock after the kernel has been started. 

Is it possible after changing the core clock (and the variable SystemCoreClock) to correct the SYSTICK simply by calling OS_Tick_Setup?  

Thanks and regards
Joachim

JonatanAntoni, 2018-10-16:
Hi Joachim,

Thank you for clarification. Changing the system clock at run-time is known to have various pitfalls.
We rather recommend to use low power and sleep modes to save power. 

If you want to use adaptive clock frequency you might update `SysTick->LOAD` and `SysTick->VAL` registers directly. This is done by the current implementation of `OS_Tick_Setup` as well. But simply calling this function might have unintended side effects. Updating the sys tick configuration appropriately might be tricky in order to keep the tick interrupt really at constant rate.

Unfortunately I cannot share practical knowledge on that very specific use case. Sharing your experience is highly appreciated. 

Cheers,
Jonatan

joabeck, 2018-10-16:
Hi Jonatan,
thank you for your advice. We will eventually valuate this.
Cheers
Joachim

closed 2018-10-16 by joabeck