ARM-software/CMSIS_5 #454: osDelay returns osOK although the scheduler is locked

MrReberT, 2018-10-19:
I am using the RTX v5 RTOS with the following version: V5.3.0.
I discovered an unexpected issue when I lock the scheduler by calling osKernelLock and the call osDelay. I expected to get an error return value from osDelay. Instead osDelay returns immediately with osOK. Is this a known bug or am I missing something here?
When I call osSemaphoreAcquire (with osWaitForever) I get an error return values as expected.
How does the RTX v5 RTOS handle calls that normally lead to a context switch when the scheduler has been disabled? In my opinion all such calls should fail and return an error return value.

JonatanAntoni, 2018-10-19:
Hi @MrReberT,

Thanks for getting in touch. You're right that calling blocking RTOS functions while having thread switching locked does not have consistent behavior. We have not considered that one calls blocking functions while in a critical region. But from error reporting point of view it looks like a fair point.

I'll discuss this issue with engineering team and report back.

Cheers,
Jonatan

RobertRostohar, 2021-05-17:
An error code should be returned now when calling osDelay/osDelayUntil while Kernel is locked.

JonatanAntoni, 2021-06-08:
@MrReberT, may I ask you to check the fix and close the issue, please?

MrReberT, 2021-06-17:
Thanks @JonatanAntoni . We are currently no longer working on the project using RTX v5 RTOS. However I will close the issue anyway.

closed 2021-06-17 by MrReberT