ARM-software/CMSIS_5 #455: feature request: nonsquare pooling functions

iabdalkader, 2018-10-19:
Hi, I have a person detection network trained on 64x128 dataset, I used the nonsquare convolution functions but the pooling functions crash because they don't support nonsquare input. I would like to request nonsquare max/ave pooling functions to work with the nonsquare convolution functions. I implemented a slow (non-dsp version) nonsquare pooling, but a dsp version will be much appreciated. Thanks!

saseptim, 2018-10-25:
As far as I can tell, the pooling function signature should be adjusted to include both x and y dimensions for pool, stride, pad, etc, and the code slightly adjusted such that dim_im_in should be changed to dim_im_in_x or dim_im_in_y as needed for example.

majianjia, 2018-12-18:
Same here. I am looking for 1D pooling, but finally, I have to make my own changes.  
All I have done is copy and paste the original one and delete all the loop with the y-axis. 
I believe you could modify it easily since the x and y are performed separately if you see the comments in the source

kwagyeman, 2018-12-18:
Would be nice in ARM fixed this.