ARM-software/CMSIS_5 #456: Compiler warning in osRtxErrorNotify()

escherstair, 2018-10-22:
While including RTOS2 API, file `RTC_config.c` contains `osRtxErrorNotify()` which gives a compilation warning:
_No return, in function returning non-void_

It's not a big issue, I know...

JonatanAntoni, 2018-10-22:
Hi @escherstair,

Thanks for making us aware of this issue. Indeed it should not be a big deal but we'll clarify this. Typically `osRtxErrorNotify()` is only called for critical and non-recoverable errors. Hence returning (a value) from this function is meaningless. 

Cheers,
Jonatan

JonatanAntoni, 2018-10-23:
Hi @escherstair,

This return value is intended for supporting future error recovery functionality. Appreciatively it has not been implemented, yet. Hence its not recommended to return from `osRtxErrorNotify()` at all, i.e. this function is to be considered `__NO_RETURN` in the current implementation.

Nevertheless we are planing to enhance the implementation someday in the future. Thus we want to keep the return value for the time being. If its feasible for you please ignore the warning.

Cheers,
Jonatan