ARM-software/CMSIS_5 #457: Event Recorder Configuration in RTX_Config.h

chentang16, 2018-10-25:
As written here http://www.keil.com/pack/doc/CMSIS/RTOS2/html/config_rtx5.html#evtrecConfig , there are two Event Filters that can be configured, one for "Global Event Filter Setup", one for "RTOS Event Filter Setup". The documentation for "Global Event Filter Setup" is clear to understand, i.e. clicking on the option means "Enable error events." or "Enable API function call events." and so on.
But the description for RTOS Event Filter Setup is a bit misleading. For instance, "Filter enable for Memory Management events." Does it mean a filter is enabled and the Memory Management events will be filtered out?
Thus, to be consistent, I suggest just writing "Enable Memory Management events.", "Enable Kernel events." and so on for "RTOS Event Filter Setup"

RobertRostohar, 2018-10-30:
Event Recorder configuration has been updated and now refers to recording levels rather than filter enable. Documentation has also been updated and should be more clear now.

JonatanAntoni, 2018-10-30:
@chentang16, please cross check the changes and close this issue if you think its solved.

closed 2018-10-31 by chentang16