ARM-software/CMSIS_5 #458: Bugs in MPU enable and disable functions in file mpu_armv8.h

MicroConsult, 2018-10-30:
Summary:
the MPU enable and disable functions in file mpu_armv8.h, which is part of the CMSIS pack 5.1.2, are faulty

Description:
* @file mpu_armv8.h
* @brief CMSIS MPU API for Armv8-M MPU
* @version V5.0.4
* @date 10. January 2018

Function in Line 115: **ARM_MPU_Enable**(uint32_t MPU_Control)
`117: __DSB();`
`118: __ISB();`
`119: MPU->CTRL = MPU_Control | MPU_CTRL_ENABLE_Msk;`
should be corrected into
`117: MPU->CTRL = MPU_Control | MPU_CTRL_ENABLE_Msk;`
`118: __DSB();`
`119: __ISB();`

Function in Line 127: **ARM_MPU_Disable**(void)
`129: __DSB();` should be changed into `129: __DMB();`
`130: __ISB();` should be deleted

For reference, please refer to document ARM 100699_0101_00_en_01
Memory Protection Unit for ARM®v8-M based platforms, Version 1.1
page 5-37 and 5-38

JonatanAntoni, 2019-03-08:
Hi @MicroConsult,

Apologies that the minor fix has taken so long.
May I ask you to cross check the change and close the issue, please?

Cheers,
Jonatan

JonatanAntoni, 2019-03-11:
@MicroConsult, can you close this issue if the provided fix is okay for you, please?

closed 2019-03-12 by MicroConsult

MicroConsult, 2019-03-12:
Hi Jonatan,
 
just short off-topic question:
In context of CMSIS for armv8-M:  Is the partition_....h header file optional?
I got used to it with the ARM_CM33 simulator model and am missing it now in context of the NXP LPC55S69.
 
Is it NXP to ask if they intend to add it to their pack or is it done on arm side?
Any chance to simple use the partition_cm33.h, include it in main and call it from there?
Any other proposal?
 
Thanks for your help.
 
Dipl. Ing. Univ.
Remo Markgraf
Senior Management Consultant

MicroConsult GmbH
Charles-de-Gaulle-Str. 6 • 81737 München
Telefon: +49 (0)89 450617-36
email: r.markgraf@microconsult.com
www.microconsult.de

 

     
     
MicroConsult  -  Microelectronics Consulting & Training GmbH     
Firmensitz:      81737 München, Charles-de-Gaulle-Str. 6     
Geschäftsführer: Ingo Pohle     
Registergericht: Amtsgericht München HRB 11 03 50     
Telefon:         089 45 06 17-71   resp.   +49 89 45 06 17 71     
Fax:             089 45 06 17-17   resp.   +49 89 45 06 17 17     
E-Mail-Kontakt:  info@microconsult.com     
Web:             http://www.microconsult.de     
                 http://www.microconsult.com

JonatanAntoni, 2019-03-12:
Hi @MicroConsult,

please consider raising new questions in their own issues rather than adding them below an uncorrelated discussion.

Short answer: The partition.h file is part of the template startup we prepared as part of CMSIS. Its not mandatory for device vendors to implement the trust zone initialization in the exact same way. Hence its up to NXP to use whatever solution fits their device best.

Cheers,
Jonatan