ARM-software/CMSIS_5 #460: resetValue tag from SVD does not end up in header

JakkoVerhallen, 2018-10-31:
Hi,

I tried SVDConv 3.3.21, both Windows and Linux, to get the reset values of registers in the header file.
My SVD file does contain the reset value:
	<name>BOD_LVL_CTRL0_REG</name>
	<addressOffset>0x00000064</addressOffset>
	<size>32</size>
	<access>read-write</access>
	**<resetValue>0x041e0eaf</resetValue>**
	<resetMask>0x07ffffff</resetMask>

But I don't see this constant value in the header file.
Is this something which can be added in a new release?
Thanks!
Jakko

JakkoVerhallen, 2018-10-31:
Corrected code:
`	<name>BOD_LVL_CTRL0_REG</name>
	
	<addressOffset>0x00000064</addressOffset>
	<size>32</size>
	<access>read-write</access>
	<resetValue>0x041e0eaf</resetValue>
	<resetMask>0x07ffffff</resetMask>
`

JonatanAntoni, 2018-10-31:
Hi @JakkoVerhallen,

Not all the information from the SVD file is used when generating the device header file. Typically only the information to access the specified registers and fields properly is put to the header.

Can you tell me a little more about your use case? What do you need the reset value for? And what type of (additional) code do you expect to be generated?

Cheers,
Jonatan

JakkoVerhallen, 2018-10-31:
Hi @JonatanAntoni,

It is useful to restore the reset value after changing the code. It might be a block which you disable, or put GPIO pads back to the original state.
The example from the SVD snippet above would be:
BOD->BOD_LVL_CTRL0_REG= 0x041e0eaf; // back to reset value
Where the 0x041e0eaf would come from the header file instead. E.g.
BOD->BOD_LVL_CTRL0_REG= **BOD_**BOD_LVL_CTRL0_REG_RESET;

The readability and intention of the second line is much more clear i.m.o.
The information currently is in SVD file, only needs to be exported to the headerfile.

Jakko

JonatanAntoni, 2018-10-31:
Thanks @JakkoVerhallen,

I'll track this as an enhancement. 

Cheers,
Jonatan

JakkoVerhallen, 2018-10-31:
Hi @JonatanAntoni ,

OK, thanks!!
Jakko