ARM-software/CMSIS_5 #461: RTX library does not build

KarolisMil, 2018-11-03:
Hi,

When I try to build customized RTX lib by project file `CMSIS_5\CMSIS\RTOS2\RTX\Library\ARM\MDK\RTX_CM.uvprojx` it does not build and says that CMSIS_device_header (ARMCM4_FP.h) is missing. What is wrong here?

JonatanAntoni, 2018-11-05:
Hi @KarolisMil,

Can you please be slightly more specific about
- how do you try to build the lib?
- which error messages you get?

I quickly cross-checked building the lib by simply using the uVisison project from my Git folder. It compiles without errors.

I suppose some include paths differs at your machine. By default the builder creates a batch script to replay the build. You could check one of the command files (.__i) to see all the include paths used. The device header should be in one of them, by default CMSIS_5/Device/ARM/ARMCM4/Include/ARMCM4_FP.h.

Cheers,
Jonatan

ReinhardKeil, 2019-01-07:
Closing the issue as we cannot duplicate the problem.

closed 2019-01-07 by ReinhardKeil