ARM-software/CMSIS_5 #462: Is it possible to create a thread in suspended state?

joabeck, 2018-11-06:
Hello,
i would like to create a thread in suspended state to be able to start (resume) the thread later.
I have not found a way to do this directly in the osThreadNew api.

Is it possible and safe to do this with the following sequence:

- osKernelLock
- osThreadNew
- osThreadSuspend (the new created thread)
- osKernelUnlock

Thank you for your support.
Best regards
Joachim

joabeck, 2018-11-11:
Hi Jonathan,
thank you for your answer. I decided to go with osThreadSuspend because that seems easier and a more logical to me.
Maybe this could be a reasonable extension to the osThreadNew API to have the possibility to create a thread directly in suspended state. For example by some new flags in the attr_bits.
Best regards
Joachim

JonatanAntoni, 2018-11-12:
Hi Joachim,

Thanks for your proposal. I'll put this as a future enhancements for now. Likely to continue discussing on this.

Cheers,
Jonatan