ARM-software/CMSIS_5 #463: Definition of __CLZ for gcc causes undefined behavior (formally)

Amomum, 2018-11-06:
For gcc `__CLZ` is defined in `cmsis_gcc.h` as follows:
``` cpp
/**
  \brief   Count leading zeros
  \details Counts the number of leading zeros of a data value.
  \param [in]  value  Value to count the leading zeros
  \return             number of leading zeros in value
 */
#define __CLZ             (uint8_t)__builtin_clz
```

According to [gcc documentation](https://gcc.gnu.org/onlinedocs/gcc/Other-Builtins.html):

> Built-in Function: int __builtin_clz (unsigned int x)
> Returns the number of leading 0-bits in x, starting at the most significant bit position. **If x is 0, the result is undefined.**

(emphasis mine)

So, strictly speaking, __builtin_clz is not guaranteed to be compiled to CLZ instruction (or is it?) and may cause UB.

I suppose that's not very important but I still would prefer to see inline assembly instead of that builtin function.

JonatanAntoni, 2018-11-07:
Hi @Amomum,

using the Compiler builtin's allows better optimization than using inline assembly. In this case the questionable part is the "undefined result", of course. I'd rather expect `__CLZ(0)` to result in 32 exactly like the CLZ instruction is defined, see http://www.keil.com/support/man/docs/armasm/armasm_dom1361289868426.htm.

At the first glance I'd consider this as a compiler issue. We can of course work around this issue in CMSIS, i.e. check for zero explicitly. Such a work around looks not clean to me. Hence we should only take it if really necessary.

Is it feasible for you to run certain tests to check if we have a problem with this builtin? If yes providing a fix is highly appreciated.

Thanks,
Jonatan

JonatanAntoni, 2018-11-07:
Hi @Amomum,

Thanks for this clarification. Of course the `__builtin_clz()` might behave differently on platforms other than Arm. We might change the implementation to something like
```
STATIC_INLINE uint8_t __CLZ(uint32_t value) {
  if(value == 0) {
    return 32U;
  }
  return __builtin_clz(value);
}
```

This might still allow the compiler to optimize CLZ but prevents a 0 operand to make it through. The downside is the code overhead. But from current GCC's spec of this built-in we would need to spent this to be clean.

Cheers,
Jonatan

kjbracey-arm, 2018-12-19:
FWIW, that form above does manage to eliminate the `if` and produce a single CLZ instruction in gcc 7.3, with -O1 or higher. (-O0 is a mess).

kjbracey-arm, 2018-12-20:
I personally would feel happier if such things were done as inline assembler if we're on an architecture that has the instruction.

I feel a little uncomfortable knowing that my "core instruction access" functions ("access to dedicated instructions") are potentially doing something different from just generating the instruction and giving the compiler the necessary optimisation directives.

The potential optimisation losses in most use cases would be minimal - only quite unusual things like being able to do that `if (0)` would seem to be gained.

If this function was called `ffs` or something, not `__<ARM_opcode_name>`, I'd not be worried at all.

But it does seem that generally you're not claiming a 1:1 mapping, and I guess if you're writing in C you know you're not in full control, so doing this seems consistent with the rest of the API implementation. I'll make a PR.

JonatanAntoni, 2018-12-20:
Hi @kjbracey-arm,

There is an ongoing discussion whether we should prefer inline assembly or rely on compiler built-ins where available. In former times we typically had inline assembly nearly everywhere due to lack off built-ins. 

We started to migrate to built-ins due to optimization requests. Having inline assembler cuts off nearly all compiler optimizations. A commonly stated example is calling `__CLZ(0x0032u)` (using a constant/quasi constant operand). Using the built-in allows the compiler to do a compiler-time calculation and remove the `__CLZ` instruction completely. Having inline assembly instead forces the compiler to keep the run time calculation.

In order to leverage compiler improvements while keeping implementations target agnostic (not manually optimized) we agreed on giving up some control.

Does this sound reasonable to you?

Thanks,
Jonatan

kjbracey-arm, 2018-12-20:
This sounds perfectly reasonable as a general principle. I think using higher-level compiler intrinsics is beneficial.

I'd just suggest that needs to be reflected in the header file and docs - that the intent of the CMSIS instrinsics is to correspond to and potentially generate code using those instructions, but they may not literally generate exactly just that instruction.

At the minute everything in the documentation suggests a very literal 1 function = 1 instruction mapping with any software fallback only based on architecture: "The following functions generate specific Cortex-M instructions that cannot be directly accessed by the C/C++ Compiler. "

(Obviously this has never been true for some of them, like `__ROR`.)