ARM-software/CMSIS_5 #464: [Question][Driver] Gpio

GuillaumeFegeFullElec, 2018-11-08:
Hi,

Apparently, no implementation of a GPIO driver is planned.
This could be useful in some library where it is necessary to : 
- read the state of an gpio input (get a state)
- set a gpio output state (launch an action)
- get an interruption/event when a gpio input change (event occured)

For example, the following library (https://github.com/Lora-net/LoRaMac-node) provide an HAL for gpio (https://github.com/Lora-net/LoRaMac-node/blob/develop/src/boards/gpio-board.h)  that could be enougth for most of our application.

Currently, we need the following features  : 
- Configure a gpio as Input or Output (with pull up / pull down / none ; Open drain / Push pull)
- read / write a gpgio state
- Configure a callback event for a gpio input change (rising / falling / rising + falling)

Is there a possibility to create such header in CMSIS driver ?

GuillaumeFegeFullElec, 2018-11-09:
Thanks for the reply,

> In the meanwhile we have put those GPIO abstractions (Buttons, LEDs, aso) to the board support packs

I do not seen any standart header abstraction for GPIO in the BSP packs. Are all BSP respect exactly the same API for GPIO ?

> Moreover just providing a CMSIS-Driver interface for GPIO is only half the job

Yes, we are aware about this. But we have the possibility to implement it for the CPU we used.  And when the pack vendors will do the job (if they do), our API will be more portable.

> Please feel free to contribute such a header. I think it is much easier to discuss the benefits based on a concrete API specification and some examples.

I'll try to make a proposal

JonatanAntoni, 2018-11-09:
Hi @GuillaumeFegeFullElec,

No, there is no standard for BSPs in that sense. Its somewhat customer specific, i.e. we defined our own board support API used by most of our MDK Middleware examples. I suspect the idea is very the same than what you are looking for, i.e. keeping the high level software application as independent as possible from the actually used device.

Cheers,
Jonatan

GuillaumeFegeFullElec, 2018-11-19:
Hi @JonatanAntoni ,
Here is a first proposal:

```
#ifndef DRIVER_GPIO_H_
#define DRIVER_GPIO_H_

#ifdef  __cplusplus
extern "C"
{
#endif

#include "Driver_Common.h"

#define ARM_GPIO_API_VERSION ARM_DRIVER_VERSION_MAJOR_MINOR(0,1)  /* API version */

/****** GPIO Pin Mode *****/
#define ARM_GPIO_PIN_MODE_IN_OUT_POS        0
#define ARM_GPIO_PIN_MODE_IN_OUT_MASK       (3UL << ARM_GPIO_PIN_MODE_INOOUT_POS)
#define ARM_GPIO_PIN_MODE_INPUT             (0UL << ARM_GPIO_PIN_MODE_INOOUT_POS)   //!< Pin as input
#define ARM_GPIO_PIN_MODE_OUTPUT_PP         (1UL << ARM_GPIO_PIN_MODE_INOOUT_POS)   //!< Pin as output push-pull
#define ARM_GPIO_PIN_MODE_OUTPUT_OD         (2UL << ARM_GPIO_PIN_MODE_INOOUT_POS)   //!< Pin as output open drain
                                                                                    // Value 3 RFU
#define ARM_GPIO_PIN_MODE_PULLUP_POS        2
#define ARM_GPIO_PIN_MODE_PULLUP_MASK       (1UL << ARM_GPIO_PIN_MODE_PULLUP_POS)
#define ARM_GPIO_PIN_MODE_PULLUP            (0UL << ARM_GPIO_PIN_MODE_PULLUP_POS)   //!< Pin with pull up
#define ARM_GPIO_PIN_MODE_NO_PULLUP         (1UL << ARM_GPIO_PIN_MODE_PULLUP_POS)   //!< Pin without pull up

#define ARM_GPIO_PIN_MODE_PULLDW_POS        3
#define ARM_GPIO_PIN_MODE_PULLDW_MASK       (1UL << ARM_GPIO_PIN_MODE_PULLDW_POS)
#define ARM_GPIO_PIN_MODE_PULLDW            (0UL << ARM_GPIO_PIN_MODE_PULLDW_POS)  //!< Pin with pull down
#define ARM_GPIO_PIN_MODE_NO_PULLDW         (1UL << ARM_GPIO_PIN_MODE_PULLDW_POS)  //!< Pin without pull down

#define ARM_GPIO_PIN_MODE_EVENT_POS         4
#define ARM_GPIO_PIN_MODE_EVENT_MASK        ( 3UL     << ARM_GPIO_PIN_MODE_EVENT_POS)
#define ARM_GPIO_PIN_MODE_EVENT_RISING      ((1UL<<1) << ARM_GPIO_PIN_MODE_EVENT_POS)   //!< Pin rising generate an event : \ref ARM_GPIO_SignalEvent
#define ARM_GPIO_PIN_MODE_EVENT_FALLING     ((1UL<<2) << ARM_GPIO_PIN_MODE_EVENT_POS)   //!< Pin falling generate an event : \ref ARM_GPIO_SignalEvent

/****** GPIO Pin State *****/
#define ARM_GPIO_PIN_STATE_LOW              0
#define ARM_GPIO_PIN_STATE_HIGH             1

/****** GPIO Pin State  *****/
#define ARM_GPIO_PIN_NUM_0_LOW                  (ARM_GPIO_PIN_STATE_HIGH << 0)   ///< GPIO PIN 0 LOW
#define ARM_GPIO_PIN_NUM_0_HIGH                 (ARM_GPIO_PIN_STATE_LOW << 0)    ///< GPIO 0 HIGH
#define ARM_GPIO_PIN_NUM_1_LOW                  (ARM_GPIO_PIN_STATE_HIGH << 1)   ///< GPIO PIN 1 LOW
#define ARM_GPIO_PIN_NUM_1_HIGH                 (ARM_GPIO_PIN_STATE_LOW << 1)    ///< GPIO 1 HIGH
#define ARM_GPIO_PIN_NUM_2_LOW                  (ARM_GPIO_PIN_STATE_HIGH << 2)   ///< GPIO PIN 2 LOW
#define ARM_GPIO_PIN_NUM_2_HIGH                 (ARM_GPIO_PIN_STATE_LOW << 2)    ///< GPIO 2 HIGH
#define ARM_GPIO_PIN_NUM_3_LOW                  (ARM_GPIO_PIN_STATE_HIGH << 3)   ///< GPIO PIN 3 LOW
#define ARM_GPIO_PIN_NUM_3_HIGH                 (ARM_GPIO_PIN_STATE_LOW << 3)    ///< GPIO 3 HIGH
#define ARM_GPIO_PIN_NUM_4_LOW                  (ARM_GPIO_PIN_STATE_HIGH << 4)   ///< GPIO PIN 4 LOW
#define ARM_GPIO_PIN_NUM_4_HIGH                 (ARM_GPIO_PIN_STATE_LOW << 4)    ///< GPIO 4 HIGH
#define ARM_GPIO_PIN_NUM_5_LOW                  (ARM_GPIO_PIN_STATE_HIGH << 5)   ///< GPIO PIN 5 LOW
#define ARM_GPIO_PIN_NUM_5_HIGH                 (ARM_GPIO_PIN_STATE_LOW << 5)    ///< GPIO 5 HIGH
#define ARM_GPIO_PIN_NUM_6_LOW                  (ARM_GPIO_PIN_STATE_HIGH << 6)   ///< GPIO PIN 6 LOW
#define ARM_GPIO_PIN_NUM_6_HIGH                 (ARM_GPIO_PIN_STATE_LOW << 6)    ///< GPIO 6 HIGH
#define ARM_GPIO_PIN_NUM_7_LOW                  (ARM_GPIO_PIN_STATE_HIGH << 7)   ///< GPIO PIN 7 LOW
#define ARM_GPIO_PIN_NUM_7_HIGH                 (ARM_GPIO_PIN_STATE_LOW << 7)    ///< GPIO 7 HIGH
#define ARM_GPIO_PIN_NUM_8_LOW                  (ARM_GPIO_PIN_STATE_HIGH << 8)   ///< GPIO PIN 8 LOW
#define ARM_GPIO_PIN_NUM_8_HIGH                 (ARM_GPIO_PIN_STATE_LOW << 8)    ///< GPIO 8 HIGH
#define ARM_GPIO_PIN_NUM_9_LOW                  (ARM_GPIO_PIN_STATE_HIGH << 9)   ///< GPIO PIN 9 LOW
#define ARM_GPIO_PIN_NUM_9_HIGH                 (ARM_GPIO_PIN_STATE_LOW << 9)    ///< GPIO 9 HIGH
#define ARM_GPIO_PIN_NUM_10_LOW                 (ARM_GPIO_PIN_STATE_HIGH << 10)  ///< GPIO PIN 10 LOW
#define ARM_GPIO_PIN_NUM_10_HIGH                (ARM_GPIO_PIN_STATE_LOW << 10)   ///< GPIO 10 HIGH
#define ARM_GPIO_PIN_NUM_11_LOW                 (ARM_GPIO_PIN_STATE_HIGH << 11)  ///< GPIO PIN 11 LOW
#define ARM_GPIO_PIN_NUM_11_HIGH                (ARM_GPIO_PIN_STATE_LOW << 11)   ///< GPIO 11 HIGH
#define ARM_GPIO_PIN_NUM_12_LOW                 (ARM_GPIO_PIN_STATE_HIGH << 12)  ///< GPIO PIN 12 LOW
#define ARM_GPIO_PIN_NUM_12_HIGH                (ARM_GPIO_PIN_STATE_LOW << 12)   ///< GPIO 12 HIGH
#define ARM_GPIO_PIN_NUM_13_LOW                 (ARM_GPIO_PIN_STATE_HIGH << 13)  ///< GPIO PIN 13 LOW
[... 144 lines cut ...]
} ARM_GPIO_CAPABILITIES;

/**
\brief Access structure of the GPIO Driver.
*/
typedef struct _ARM_DRIVER_GPIO {
  ARM_DRIVER_VERSION    (*GetVersion)      (void);                                    ///< Pointer to \ref ARM_GPIO_GetVersion : Get driver version.
  ARM_GPIO_CAPABILITIES (*GetCapabilities) (void);                                    ///< Pointer to \ref ARM_GPIO_GetCapabilities : Get driver capabilities.
  int32_t               (*Initialize)      (ARM_GPIO_SignalEvent_t cb_event);         ///< Pointer to \ref ARM_GPIO_Initialize : Initialize GPIO Interface.
[... cut ...]
  int32_t               (*GetPin)          (uint32_t pin_num);                        ///< Pointer to \ref ARM_GPIO_GetPin : Get pin state
  int32_t               (*SetPort)         (uint32_t pin_mask, uint32_t state_mask);  ///< Pointer to \ref ARM_GPIO_SetPort : Set multiple pin state
  int32_t               (*GetPort)         (uint32_t pin_mask);                       ///< Pointer to \ref ARM_GPIO_GetPort : Get multiple pin state
  
} const ARM_DRIVER_GPIO;

#ifdef  __cplusplus
}
#endif

#endif /* DRIVER_GPIO_H_ */
```

mishka88ru, 2019-01-17:
Hello :)

Nice! :+1: 

But why not like this?
```
/****** GPIO Pin State  *****/
#define ARM_GPIO_PIN_NUM_LOW (n)   (ARM_GPIO_PIN_STATE_HIGH << (n))   ///< GPIO PIN n LOW
#define ARM_GPIO_PIN_NUM_HIGH(n)   (ARM_GPIO_PIN_STATE_LOW  << (n))   ///< GPIO n HIGH

/****** GPIO Pin Number *****/
#define ARM_GPIO_PIN_NUM_MASK(n)   (1UL << (n))   ///< GPIO PIN Number/Mask for pin n  
```

**Or let SetPin to accept pin_num as number in [0;31]. Then we don't even need _NUM_LOW/_NUM_HIGH.**

By the way, NXP Kinetis provide special register to toggle output:
```
SetPin       (uint32_t pin_num);                    // write 1
ClearPin     (uint32_t pin_num);                    // write 0
TogglePin    (uint32_t pin_num);                    // toggle current value
WritePin     (uint32_t pin_num, uint32_t value);    // write value 0/1
ReadPin      (uint32_t pin_num);                    // return current value

SetPort       (uint32_t pin_mask);   // write 1s if pin_mask[i] == 1
ClearPort     (uint32_t pin_mask);   // write 0s if pin_mask[i] == 1
TogglePort    (uint32_t pin_mask);   // toggle current value if pin_mask[i] == 1
ReadPort      ();                    // return current values
```

And more events:
* Interrupt when logic 0
* Interrupt on rising-edge
* Interrupt on falling-edge
* Interrupt on rising-edge
* Interrupt when logic 1

JonatanAntoni, 2019-01-17:
Hi @mishka88ru,

thanks for contributing to this discussion. You exactly hit the point why we have not defined a GPIO driver API in CMSIS so far. Different silicon vendors offer different features for their devices.

Offering a generic API would always end up in some trade-offs, e.g. performance vs. ease-of-use or simplicity vs. completeness.

Finally defining the API is only half the way. Someone needs to implement the API for all the devices ;)

Cheers,
Jonatan

GuillaumeFegeFullElec, 2019-01-17:
> Hello :)
> 
> Nice! 
>
Thanks :-) 

> But why not like this?
>
> ```
> /****** GPIO Pin State  *****/
> #define ARM_GPIO_PIN_NUM_LOW (n)   (ARM_GPIO_PIN_STATE_HIGH << (n))   ///< GPIO PIN n LOW
> #define ARM_GPIO_PIN_NUM_HIGH(n) [...]
Ok for me,

> ```
> **Or let SetPin to accept pin_num as number in [0;31]. Then we don't even need _NUM_LOW/_NUM_HIGH.**
The aim was to be able to change multiple pin in the same time

> By the way, NXP Kinetis provide special register to toggle output:
>
> ```
> SetPin       (uint32_t pin_num);                    // write 1
> ClearPin     (uint32_t pin_num);                    // write 0 [...]
So for NXP, the SetPort and ClearPort could be used.

> ```
> And more events:
> 
> * Interrupt when logic 0
> * Interrupt on rising-edge
> * Interrupt on falling-edge
> * Interrupt on rising-edge
> * Interrupt when logic 1
The events when logic 0 or when logic is 1 doesn't make sens for me. Cause it is not an event but a state. So the interrutpion will be called undefinetly

GuillaumeFegeFullElec, 2019-01-17:
> thanks for contributing to this discussion. You exactly hit the point why we have not defined a GPIO driver API in CMSIS so far. Different silicon vendors offer different features for their [...]

Yes, some functions will be optimize for some manufacturer, and not for other. But it is the same for all driver (SPI, I2C, ...)
 
> Finally defining the API is only half the way. Someone needs to implement the API for all the devices ;)

Yes, but before implement it, we have to specify it, no ?

mishka88ru, 2019-01-17:
> The events when logic 0 or when logic is 1 doesn't make sens for me. Cause it is not an event but a state. So the interrutpion will be called undefinetly

To be honest, I don't use this feature. Found some theory [here](https://electronics.stackexchange.com/a/175946).

> > Finally defining the API is only half the way. Someone needs to implement the API for all the devices ;)

> Yes, but before implement it, we have to specify it, no ?

Some minimalistic API can be a good start. GPIO is not rocket science :)  
Things like toggle can be easily done in sw: `write(n, 1 - read(n))`. I do not insist on toggle - it can be done not in a driver; I just thought it's pretty useful and should be coded somewhere anyway (why not close to metal...)

JonatanAntoni, 2019-01-17:
Please go ahead. I'd suggest you both prepare a PR with an API implementation that fits your both needs. As a proof of concept I'd like to see reference implementations for at least two devices from different vendors. Can you give us some examples how this would look like?

Adding this one should be fairly simple. But bear in mind that changing the API after people having started to implement it will be cumbersome.

BTW: Level based interrupts might make sense, e.g. for external interrupt lines. Devices tend to keep the interrupt line pulled low until the request has been cleared.

GuillaumeFegeFullElec, 2019-01-17:
> > The events when logic 0 or when logic is 1 doesn't make sens for me. Cause it is not an event but a state. So the interrutpion will be called undefinetly
> 
> To be honest, I don't use this feature. Found some theory [here](https://electronics.stackexchange.com/a/175946).

OK, i understand te point.

> Some minimalistic API can be a good start. GPIO is not rocket science :)
> Things like toggle can be easily done in sw: `write(n, 1 - read(n))`. I do not insist on toggle - it can be done not in a [...]

Ok for toogle (I forgot it) 

>Please go ahead. I'd suggest you both prepare a PR with an API implementation that fits your both >needs. As a proof of concept I'd like to see reference implementations for at least two devices from >different vendors. Can you give us some examples how this would look like?

Ok, i'll try to do it. I could implement it for an STMicroelectronics device.

mishka88ru, 2019-01-21:
Experimental implementation for NXP Kinetis K66 [https://github.com/mishka88ru/arm-cmsis-gpio](https://github.com/mishka88ru/arm-cmsis-gpio)

> But bear in mind that changing the API after people having started to implement it will be cumbersome.

I am ready for questions, critique, suggestions, etc...

JonatanAntoni, 2019-01-22:
@RobertRostohar may I ask you to give this a review, please?