ARM-software/CMSIS_5 #465: RTE_components.h missing

KarolisMil, 2018-11-11:
Hello,
I am building using GCC (using library, not form source) and with new version (5.4.0) I get an error:
```
In file included from src/cmsis/rtos2/RTX/Source/rtx_lib.c:34:0:
src/cmsis/rtos2/RTX/Include/rtx_evr.h:33:28: fatal error: RTE_Components.h: No such file or directory
 #include "RTE_Components.h"
                            ^
compilation terminated.
```
I do not have any RTE_components.h since I am not building with Keil. Can you provide a solution? Also, I do not use event recorder.

JonatanAntoni, 2018-11-12:
Hi @KarolisMil,

This looks like a side effect when using RTX5 outside an RTE. As a quick fix you can simply create an empty `RTE_Components.h` somewhere in your include path. I need some more time to investigate how we can fix this properly.

Thanks,
Jonatan

KarolisMil, 2018-11-12:
Do you have any plans in your roadmap to make CMSIS out-of-the-box compatible with make projects without using Keil or DS studio?

KarolisMil, 2018-11-12:
Can I find any documentation how to generate (or write by hand) `RTE_components.h` file?

Also, but in case user like me uses GCC without any IDE, and needs to build RTX lib, it is needed to use... Keil, not a portable option like make. And user has to write Makefile by his/her own which is spot of failure and bugs. Do you have plans to get rid of IDE's completely to be as portable as possible?

JonatanAntoni, 2018-11-12:
Hi @KarolisMil,

the `RTE_Components.h` is documented here: https://www.keil.com/pack/doc/CMSIS/Pack/html/pdsc_components_pg.html#RTE_Components_h

I admit that we don't support make currently out-of-the-box. All you need to know about a software component to build (or use) it properly should be in the pack description, e.g. the component definition of RTX5 at https://github.com/ARM-software/CMSIS_5/blob/1ea17c52c965b2a1a0086bcd414a7429b36802a0/ARM.CMSIS.pdsc#L3372. 

The CMSIS Pack concept is how we made software components portable. Its completely technology and vendor agnostic. Currently its supported by a range of IDEs (including MDK and IAR) and used by dozens of (silicon) vendors to ship their software components.

Cheers,
Jonatan

JonatanAntoni, 2019-03-11:
Hi  @KarolisMil,

the issue should be fixed since 

https://github.com/ARM-software/CMSIS_5/commit/856be7008f7b22aa7f6f8a05486f0c7ab9003398

Can you cross check the solution and close the issue, please?

Thanks,
Jonatan

JonatanAntoni, 2019-05-21:
Closing due to no further activity.

closed 2019-05-21 by JonatanAntoni

Seppeltronics, 2020-08-18:
Hello Mr. Antoni,

well, I followed your recommendation and **added an empty RTE_Components.h file.** What happened next is that it fails when compiling the main_s.c  .

**Error: ../CMSIS/Target/main_s.c:29:10: error: #include expects "FILENAME" or <FILENAME>**

Reason, the #include CMSIS_device_header is defined in RTE_Components.h, which documents in “Getting started with MDK” Page 51 ( https://www.mouser.com/datasheet/2/211/dk5-getting-started-768690.pdf ).

I do not how that should be resolved and which header should be included.

In conclusion, you seem to be wrong and the CMSIS can not be compiled without modifications. Modifications that are proprietary and undocumented. What should be done is resolving this by removing the Keil-Code from the ARM-Generic code, or as a “Hack” have proper compile-switches for the Keil-Code.

Could you please at least try it yourself and tell me **how that can be resolved?**
Thank you very much.

Best Regards, Seppel

JonatanAntoni, 2020-08-31:
Hi @Seppeltronics,

I think this should be better discussed in a new issue. Can you please open a new issue and briefly describe what you are doing? Which project do you compile? Which tool environment are you using?

Thanks,
Jonatan