ARM-software/CMSIS_5 #469: feature request: add cpu init hook in very beginning of Reset_Handler

alfedotov, 2018-11-23:
There are Cortex-A/M powered chips which contains special features e.g. ECC memory that in turn must be initialized before any access.
Is it possible to add kind of cpu_init_hook in a very beginning of Reset_Handler ?

alfedotov, 2018-11-26:
Hi @JonatanAntoni 
Than you, I've seen this function. Unfortunately it is too late because right before `SystemInit` there are loops that makes access to the RAM which in turn not initialized yet.

JonatanAntoni, 2018-11-26:
Hi @alfedotov,

`SystemInit` should be the very first function called during `ResetHandler`.

Which device are you using? Where does the implementation of the `ResetHandler` come from? I suspect its not one of our CMSIS templates.

From CMSIS point of view the device startup code (include `ResetHandler`) is intended to be adopted by the user. You might want to move the call to `SystemInit` to the very beginning of `ResetHandler`.

Cheers,
Jonatan

alfedotov, 2018-11-26:
@JonatanAntoni 
Yes, I'm referring on GCC for M33 for instance. I see now that you have different `Reset_Handler` behavior on GCC/ARM/IAR compilers. The latter uses library implementation for ROM2RAM and BBS init).
In this case following your propose to utilize `SystemInit` latter must be hoisted (before ROM2RAM and BSS loops)

JonatanAntoni, 2018-11-26:
Hi @alfedotov,

Yes, I admit we are out-of-sync between different compilers. I think we need to update the GCC startup code templates, soon. `SystemInit` was introduced to do exactly what you are looking for: Setting up clocks, bus system and external memories prior to transfer any data. 

Cheers,
Jonatan

alfedotov, 2018-11-26:
@JonatanAntoni 
Please note that `SystemInit` must have attribute 'naked' to not use the Stack as well `Reset_Handler`:

```
void __attribute__((naked)) Reset_Handler(void) {
  register uint32_t *pSrc, *pDest;
  register uint32_t *pTable __attribute__((unused));
...
```
Original code here [here](https://github.com/ARM-software/CMSIS_5/blob/a540b2a37ac3aa63c42f71130cda48f8d5537d29/Device/ARM/ARMCM33/Source/GCC/startup_ARMCM33.c#L162).

alfedotov, 2018-11-27:
There is existing issue that is related https://github.com/ARM-software/CMSIS_5/issues/405

JonatanAntoni, 2019-03-11:
Closing as dup for #405. 
@alfedotov, may I ask you to verify the provided fix, please? If you see any open issue to be discussed please comment next to #405. Thanks.

closed 2019-03-11 by JonatanAntoni

alfedotov, 2019-03-13:
@JonatanAntoni  Seems fine to me as I did that fix :)