ARM-software/CMSIS_5 #474: Q15 RMS function works poorly with small numbers

d0d0-the-bird, 2018-12-02:
Hi,

the issue I am having is with the following function:
`void arm_rms_q15 (q15_t *pSrc, uint32_t blockSize, q15_t *pResult)`
If you have mainly small numbers in array pSrc you will get a very bad result back, due to the rounding errors. This rounding error is introduced in the very last part of function execution, and can be easily avoided.
To give you an example of an error, lets say you pass an array of 10 `q15_t` to `arm_rms_q15` function. Every element of this array is equal to 400 decimal. As a result of the RMS function execution you would get a value of 362 decimal. The same happens if array has only one element.
To avoid this behaviour higher resolution SQRT function could be used instead of the current solution.
Current solution in the file [https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/StatisticsFunctions/arm_rms_q15.c#L107](arm_rms_q15.c,line107) looks like the following:
` 
/* Truncating and saturating the accumulator to 1.15 format */
/* Store the result in the destination */
arm_sqrt_q15(__SSAT((sum / (q63_t)blockSize) >> 15, 16), pResult);
`
To decrease the rounding error it could be simply replaced with the following:
` 
/* Truncating and saturating the accumulator from 34.30 to 1.31 format */
/* Store the result in the temporary out */
arm_sqrt_q31(__SSAT((sum / (q63_t)blockSize) << 1, 32), &out);

/* Truncating and saturating the temporary out from 1.31 to 1.15 format */
*pResult = __SSAT(out >> 16, 16);
`

llefaucheur, 2018-12-03:
Hi, it seems the issue is related to the division, and the current implementation may have to trade between accuracy and complexity increase using sqrt_q31. The first patch consists to add to the numerator   the value (denominator/2) before division. This will round the computation before square-root. Going forward you noticed the sqrt implementation is using a float32 which is not the best idea. Next releases will use a classical Newton instead.

d0d0-the-bird, 2018-12-08:
Thank you for your quick reply! Sure, adding denominator/2 would round the result, but that is not what I want, or what I was talking about. Maybe I wasn't clear enough in the first post.

It didn't seem to be such a high increase in complexity to use sqrt_q31, that is why I proposed it.
In my use case I absolutely have to use sqrt_q31 because I need that accuracy at low numbers, but now I understand why you went with the sqrt_q15 instead.

Just out of curiosity, do you know how much longer does it take to compute sqrt_q31 vs sqrt_q15?

llefaucheur, 2018-12-08:
Hello, just an other advice, you may want to consider float32 when accuracy is a must. An other usual trick is to insert a test for low numbers and rescale by a power of 2 (2^8 for example) before calling the square root. Regards

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

closed 2019-07-30 by JonatanAntoni