ARM-software/CMSIS_5 #477: Add 'fraction saving' to DSP fixed point cascaded biquad functions

davydnorris, 2018-12-09:
I have just started using the CMSIS DSP library for my own ESP8266 based embedded project, and noticed that the biquad filters do not use the technique that's sometimes called 'fraction saving', also known as DC blocking with noise shaping. Basically you store the bits left over after you shift the accumulator and put them back into it. 

This adds one more state variable to the pState array but you end up getting an output from arm_biquad_cascade_df1_q31() that's indistinguishable from the arm_biquad_cas_df1_32x64_q31() high precision filter, but for almost half the state storage and in my tests on a 48KHz sample I shaved 200ms per second off each second due to the decrease in 64bit long manipulation.

This is the adjusted generic section of arm_biquad_cascade_df1_q31() function - the same technique could be used in all the other versions as well. I've marked the changes with the comment /* FRACTION SAVING*/:

```
void arm_biquad_cascade_df1_q31(
  const arm_biquad_casd_df1_inst_q31 * S,
  q31_t * pSrc,
  q31_t * pDst,
  uint32_t blockSize)
{
  q63_t acc;                                     /*  accumulator                   */
  uint32_t uShift = ((uint32_t) S->postShift + 1U);
  uint32_t lShift = 32U - uShift;                /*  Shift to be applied to the output */
  uint32_t mask = (1<<lShift)-1;      /* FRACTION SAVING */
  q31_t *pIn = pSrc;                             /*  input pointer initialization  */
  q31_t *pOut = pDst;                            /*  output pointer initialization */
  q31_t *pState = S->pState;                     /*  pState pointer initialization */
  q31_t *pCoeffs = S->pCoeffs;                   /*  coeff pointer initialization  */
  q31_t Xn1, Xn2, Yn1, Yn2;                      /*  Filter state variables        */
  q31_t b0, b1, b2, a1, a2;                      /*  Filter coefficients           */
  q31_t Xn;                                      /*  temporary input               */
  uint32_t sample, stage = S->numStages;         /*  loop counters                     */

  /* Run the below code for Cortex-M0 */

  do
  {
    /* Reading the coefficients */
    b0 = *pCoeffs++;
    b1 = *pCoeffs++;
    b2 = *pCoeffs++;
    a1 = *pCoeffs++;
    a2 = *pCoeffs++;

    /* Reading the state values */
    Xn1 = pState[0];
    Xn2 = pState[1];
    Yn1 = pState[2];
    Yn2 = pState[3];
    acc = (q63_t) pState[4];    /* FRACTION SAVING */

    /*      The variables acc holds the output value that is computed:
     *    acc =  b0 * x[n] + b1 * x[n-1] + b2 * x[n-2] + a1 * y[n-1] + a2 * y[n-2]
     */

    sample = blockSize;

    while (sample > 0U)
    {
      /* Read the input */
      Xn = *pIn++;

      /* acc =  b0 * x[n] + b1 * x[n-1] + b2 * x[n-2] + a1 * y[n-1] + a2 * y[n-2] */
      /* acc =  b0 * x[n] */
      acc += (q63_t) b0 *Xn;    /* FRACTION SAVING */

      /* acc +=  b1 * x[n-1] */
      acc += (q63_t) b1 *Xn1;
      /* acc +=  b[2] * x[n-2] */
      acc += (q63_t) b2 *Xn2;
      /* acc +=  a1 * y[n-1] */
      acc += (q63_t) a1 *Yn1;
      /* acc +=  a2 * y[n-2] */
      acc += (q63_t) a2 *Yn2;

      /* The result is converted to 1.31  */
      /* acc = acc >> lShift; */    /* FRACTION SAVING */

      /* Every time after the output is computed state should be updated. */
      /* The states should be updated as:  */
      /* Xn2 = Xn1    */
      /* Xn1 = Xn     */
      /* Yn2 = Yn1    */
      /* Yn1 = acc    */
      Xn2 = Xn1;
      Xn1 = Xn;
      Yn2 = Yn1;
      Yn1 = (q31_t) (acc>> lShift);    /* FRACTION SAVING */
      acc &= mask;      /* FRACTION SAVING */

      /* Store the output in the destination buffer. */
      *pOut++ = Yn1;    /* FRACTION SAVING */

      /* decrement the loop counter */
      sample--;
    }

    /*  The first stage goes from the input buffer to the output buffer. */
    /*  Subsequent stages occur in-place in the output buffer */
    pIn = pDst;

    /* Reset to destination pointer */
    pOut = pDst;

    /*  Store the updated state variables back into the pState array */
    *pState++ = Xn1;
    *pState++ = Xn2;
    *pState++ = Yn1;
    *pState++ = Yn2;
    *pState++ = (q31_t)acc;    /* FRACTION SAVING */

  } while (--stage);
}

```

davydnorris, 2018-12-10:
One other thing - the other thing this modification does is completely eliminate any DC offset generated by the filter. I just ran the code with one of my coefficient sets and it performed as well as the high precision, but was the only one that completely zeroed out like the floating point version. The rest had a DC offset (and the fast implementation had a noisy DC offset)

christophe0606, 2019-05-29:
It is a good feedback. We need to look at it.

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606