ARM-software/CMSIS_5 #479: Calculation of memory pool block size as part of the API

embeddedteam103, 2018-12-17:
Hello,
We have the following situation:
We wrapped the `threadx` closed source operating system with the `cmsis` api to be used as part of `mbed`.
This is completely closed source unfortunately as we are not using only public `tx_api.h` functions in our cmsis wrapper (maybe we could if the resources are allocated, and open source the whole situation).

The problem is in some cases, internal `RTX` details are used in `mbed` directly (in an implicit way) and NOT the `cmsis` abstraction layer which caused us to fork `mbed`.
We do not want this situation to continue, so we want to propose extensions to the public API of `cmsis` to be used in `mbed` instead of internal `RTX` "details".
We really want to use the official repos in our project and changes to `cmsis` and `mbed` are required.

The simplest of these changes is brought here in this issue:

A feature request to add a macro that calculates the size of a memory pool block size for a given type.

The code we forked in `mbed` is the `MemoryPool` class template's member `_pool_mem` (line 119 in `rtos\MemoryPool.h`):

In `threadx` every memory block requires an extra pointer of internal use (padding, from the user's perspective), therefore our fork looks like this:

`#if defined(MBED_OS_BACCKEND_THREADX)
#define CALCULATE_POOL_BLOCK_SIZE_PADDING(DATA_SIZE) (DATA_SIZE + sizeof(void*))
#else
#define CALCULATE_POOL_BLOCK_SIZE_PADDING(DATA_SIZE) (DATA_SIZE)
#endif
char _pool_mem[((CALCULATE_POOL_BLOCK_SIZE_PADDING(sizeof(T)) + 3) & ~3) * pool_sz];
#undef CALCULATE_POOL_BLOCK_SIZE_PADDING
`

We propose to add a similar macro to the cmsis API (and a trivial implementation in `RTX` of course)
Is this extension acceptable? If so we will post PRs to `cmsis`, `RTX` and `mbed-os` to propose the extensions needed for the new API.

Do you think such a change is acceptable?

0xc0170, 2018-12-17:
Hi @embeddedteam103, thanks for the report.

> We do not want this situation to continue, so we want to propose extensions to the public API of cmsis to be used in mbed instead of internal RTX "details".

This would be ideal 👍 

Did you identify any other extensions needed besides the mempool block? We would like to know and address these.
I'll create an issue on Mbed OS to request change to address this, we can continue there (fix the current situation as it is, apart from the proposal to extend CMSIS API). 
`mbed_rtos_storage.h` could have a define forosRtxMemoryPoolMemSize that this mem_pool uses.  I assume you redefine `mbed_rtos_storage.h` header file so would not need a change in MemoryPool class.

JonatanAntoni, 2018-12-17:
Hi @embeddedteam103,

I already discussed this issue with a colleague from mbed team. The macro you are looking for is already available from RTX point of view, see [osRtxMemoryPoolMemSize](https://github.com/ARM-software/CMSIS_5/blob/19cdfbf6b8a81cbdf627cc3059ddf5c4c5a91402/CMSIS/RTOS2/RTX/Include/rtx_os.h#L382). Unfortunately it is currently not used in mbed.

Currently all the sizing macros are implementation specific and not available on CMSIS-RTOS2 API. We discussed whether its viable to add a similar thing to the abstract API. We have not come to a conclusion, yet. Hence mbed contains additional abstraction of implementation specifics.

Cheers,
Jonatan

embeddedteam103, 2018-12-17:
> Did you identify any other extensions needed besides the mempool block? We would like to know and address these.

Yes, I will post these as separate issues in the mbed-os repo.

ReinhardKeil, 2019-01-09:
Can we close this issue a the solution is provided?

JonatanAntoni, 2019-03-11:
Closed due to inactivity.

closed 2019-03-11 by JonatanAntoni