ARM-software/CMSIS_5 #480: Undefined reference to arm_mat_trans_f32

guglie, 2018-12-17:
Sorry, simple question here, I've followed the usage instructions.
I have included `DSP/Include/arm_math.h` and linked `Lib/GCC/libarm_cortexM4lf_math.a`, but the compiler gives this error:
**Undefined reference to arm_mat_trans_f32**

Am I missing something?

Thanks

JonatanAntoni, 2018-12-17:
Hi @guglie,

We cannot reproduce your issue here.

```
arm-none-eabi-objdump -t libarm_cortexM4lf_math.a  | grep arm_mat_trans_f32
arm_mat_trans_f32.o:     file format elf32-littlearm
00000000 l    df *ABS*  00000000 arm_mat_trans_f32.c
00000000 l    d  .text.arm_mat_trans_f32        00000000 .text.arm_mat_trans_f32
00000000 g     F .text.arm_mat_trans_f32        000000e6 arm_mat_trans_f32
```

The symbol seems to be part of that library file. Can you please give us slightly more details about your issue and the environment your are running?

Cheers,
Jonatan

guglie, 2018-12-17:
Thank you @JonatanAntoni, there was a problem in the linker setup of my project (using gcc with waf).
Now I have succeded in compiling.

By the way can you confirm that to use the library I only need to
* include **arm_math.h**
* link to the right static library (e.g. **libarm_cortexM4l_math.a**)
* define the related macro (e.g. **ARM_MATH_CM4**)
right?

To link `-Lpath/to/CMSIS/CMSIS/Lib/GCC/ -llibarm_cortexM4l_math`, right?

GuentherMartin, 2018-12-18:
@guglie 
Yes, this is correct.

Please be aware that with the latest changes of DSP the setting of the related macro e.g. ARM_MATH_CM4 is obsolete. But the related static libraries are not yet committed. The old binaries should also work.

closed 2019-07-30 by JonatanAntoni