ARM-software/CMSIS_5 #483: RTOS2 Api, thread waiting for different objects at the same time

joabeck, 2018-12-19:


joabeck, 2018-12-19:
Hello,
i want to have a thread which can wait for an event flag or thread flag **and** for a message queue at the same time. The thread should continue, if either the flag is set or a message is in the queue. In the API i see no possibility to achieve this.
Is there any way to do this?
Thank you for your support.
Best regards
Joachim

JonatanAntoni, 2018-12-20:
Hi @joabeck,

Unfortunately this is currently not built-in to the API. A thread can only be blocked waiting on a single resource (plus timeout). To achieve something similar you would need to wrap the native RTOS objects into a custom high-level object. A possible solution would be using thread flags to signal what actually happened to the thread. You need to accomplish that the according thread flag is set whenever one of your events happen.

Cheers,
Jonatan

joabeck, 2018-12-20:
Hi Jonatan,
thank you for your answer. 
Can you please explain what you mean with "to wrap the native RTOS objects into a custom high-level object"? How can this work without support of the RTOS?
I think, i understand what you mean with the event flags. But that implies that i cannot use any other RTOS functions like queues etc., or i need a helper thread for each event (EventFlag, Queue etc.) i want to wait for in my worker thread.
Thank you

Cheers
Joachim

JonatanAntoni, 2018-12-20:
Hi Joachim,

The solution I have in mind is more like "a work around".

Lets say Thread A want to wait for either a message on a queue or an event flag. Instead of blocking on the message queue or the event flags object directly you need to block on the thread flags. 
Lets assume Thread B sends messages. On every new message it additionally sets a specific thread flag to Thread A to wake it. Same applies to Thread C when sending event flags. If you assign a dedicated thread flag to all wake up sources Thread A can directly check the according channel.

Does this makes sense to you?

Cheers,
Jonatan

joabeck, 2018-12-20:
Hi Jonatan,
ok, I understand this, but this makes things a little bit uncomfortable.
Maybe you can checks this, if it could be a possible feature for future enhancements to the API. 
Thank you
Cheers
Joachim

closed 2018-12-20 by joabeck