ARM-software/CMSIS_5 #488: API is very type unsafe

ensc, 2018-12-21:
Actual API is very type unsafe because underlying opaque types are
simple 'void *' pointers and nearly everything can be casted implicitly
to it.

E.g.

| int a;
|
| osThreadFlagsSet(&a, 1);

will be accepted by C compilers with strictest warnings levels.

API should be changed to more strict types and all the

| typedef void * ...._t;

should be removed/replaced.

JonatanAntoni, 2018-12-21:
Hi @ensc,

Do you have an idea how to strongly type the APIs while keeping them implementation agnostic? I.e. using FreeRTOS one needs to call osThreadFlagSet on a FreeRTOS-specific thread object while the same applies to RTX5 with an RTX-specific one. 

Actually the first parameter to all the CMSIS-RTOS2 functions are strictly speaking only "object ids". Hence its up to the implementation to check it for validity. The drawback if this decoupling is reduced compiler diagnostics, i.e. one does not get warnings when passing invalid object ids. This decoupling is needed to implement generic middleware components not depending on a specific RTOS implementation.

Cheers,
Jonatan

ensc, 2018-12-21:
Extend the API requirements so that each implementation has to provide an (opaque) datatype for these objects.  E.g.

```c
struct osThreadImpl;
typedef struct osThreadImpl *osEventFlagsId_t;
```

in the public API header and

```c
struct osThreadImpl {
    uint8_t                  id;
    ...
};
```

in the implementation A and

```c
struct osThreadImpl {
    char const *             name;
    unsigned long            flags;
    ...
};
```

in implementation B.

JonatanAntoni, 2019-03-11:
@RobertRostohar, can you comment on this, please?

RobertRostohar, 2019-03-13:
The proposed approach is interesting and we will explore it further. We need to see if there are other side-effects and also check MISRA compliance.

legier, 2019-09-28:
I have the same feeling as reporter of this feature.
In certain structers API keeps `void*` and while using API's functions it clearly expects this `void*` to be pointers of some type. Why not change struct implementation from `void*`to pointer of expected type? Here is example:
```C++
/// Attributes structure for mutex.
typedef struct {
  const char                   *name;   ///< name of the mutex
  uint32_t                 attr_bits;   ///< attribute bits
  void                      *cb_mem;    ///< memory for control block
  uint32_t                   cb_size;   ///< size of provided memory for control block
} osMutexAttr_t;
```

`cb_mem` in API's functions is expected to be `StaticSemaphore_t*`. Why not put it there in the first place? Like this:
```C++
/// Attributes structure for mutex.
typedef struct {
  const char                   *name;   ///< name of the mutex
  uint32_t                 attr_bits;   ///< attribute bits
  StaticSemaphore_t         *cb_mem;    ///< memory for control block
  uint32_t                   cb_size;   ///< size of provided memory for control block
} osMutexAttr_t;
```

Such code whould be much cleaner and leads to less amount of wrong usages. And because of fact that this struct is only used for mutexes (similar feature for almost all such structures) it should not be a poreblem to change it.

JonatanAntoni, 2019-10-01:
Hi @legier,

basically being more type-explicit it a very good choice. In you concrete example I don't see a clear solution.

The type `osMutexAttr_t` is defined on CMSIS-RTOS2 API level, which is intended to be implementation agnostic, i.e. shall be adoptable to RTX5 as well as to FreeRTOS or others. In contrast the usage of `memory for the control block` is implementation specific. From API point of view this is only an anonymous block of memory with a size given by `cb_size`.

The design limitation gets visible if you think about the following scenario: The vendor of an arbitrary RTOS wants to ship pre-built libraries. The user wants to link those to his specific application. Strictly speaking, both don't know about each other at compile time. They only agree on CMSIS-RTOS2 APIs. Now: What would be the API-generic declaration of `StaticSemaphore_t`? Without taking implementation specific knowledge into account, i.e. `sizeof(StaticSemaphore_t)` is unknown?

Cheers,
Jonatan

legier, 2019-10-01:
I can agree with you that created code should not be implementation depended, but in my opinion this implementation doesn't provide necessery mechanisms to create implementation independed code.

In your example there is no way to use CMSIS-RTOS2 APIs without any knowlage of what is deeper. `StaticSemaphore_t` (or any struct which should be placed there) does only need to be forward declared on API level and defined inside used system. API can provide its own type e.g. `osStaticSemaphore_t` and API's implementation related to specific system will provide suitable `typedef` and also its `sizeof`.

JonatanAntoni, 2019-10-01:
Yes, that's what I think @ensc already proposed, earlier. Changing APIs might break backward compatibility. Hence we cannot easily do these changes. Thus I flagged the idea as "future". I appreciate your vote for that enhancement.

ensc, 2019-10-01:
@JonatanAntoni   you do not need to know `sizeof(StaticSemaphore_t)` but  `sizeof(StaticSemaphore_t *)`.   The C Standard requires

> All pointers to structure types shall have the same representation and alignment requirements as each other

so that the `void *` can be replaced by a pointer to a more type safe implementation when its underlying type is a `struct`.

But the `struct` constraint matters only on esoteric/theoretical platforms (and would break the actual implementation because `sizeof(void *) < sizeof(StaticSemaphore_t *)` might hold there).