ARM-software/CMSIS_5 #489: How can i run this examples?

babatn1, 2018-12-23:
Hi everyone;

I am a new student studying CMSIS-NN, I want to run these examples, Who can tutorial help me to do this?

Many thanks!

lishunde, 2018-12-23:
you must need a Keil on Windows or ambedcli on Linux fristly！

babatn1, 2018-12-24:
Dear @lishunde ;

I  installed Keil on Windows. What are the next steps?

rarazac, 2018-12-29:
We have developed a simple parser which simplifies the work flow with the CMSIS-NN:

https://github.com/InES-HPMM/k2arm/

To try it out, you can simply copy the CMSIS-NN related files, which can be generated with our parser:
- [fully_connected.c](https://github.com/InES-HPMM/k2arm/blob/master/target/Src/fully_connected.c)
- [fully_connected.h](https://github.com/InES-HPMM/k2arm/blob/master/target/Inc/fully_connected.h)
- [weights.h](https://github.com/InES-HPMM/k2arm/blob/master/target/Inc/weights.h)

Into your Keil project and try it out. Make sure that the CMSIS-NN is included in your project.

You can also generate own nets using the python script [main.py](https://github.com/InES-HPMM/k2arm/blob/master/host/main.py) because the default net is just a MNIST classifier, you can adapt the keras model to your needs.

rarazac, 2018-12-29:
The parser is very simple and generates plain C code, given a (restricted) keras model. The C code depends only on the CMSIS-NN library, therefore I think you can use it for your project aswell. I used a STM32F429 discovery board for the example project, but you could use any MC which supports the CMSIS-NN.

kwagyeman, 2018-12-29:
Okay, it shouldn't be too much trouble to edit the code to target how tour loadable format works given we already output to CMSIS-NN calls.

Note that what I'm talking about is this: https://github.com/openmv/openmv/tree/master/ml/cmsisnn

It's a dynamic loader environment. So, the model is not forced to be compiled in but can be loaded, run, and freed dynamically.

babatn1, 2019-01-14:
Dear @lishunde, @kwagyeman , @rarazac, @JonatanAntoni ; 

I very thank everyone; I am trying to do it as your tutorial. I have a question;
So, Is there any connection between CMSIS_5 and Compiler? Please, give me an idea or any document about it!
Thank again!

majianjia, 2019-02-12:
> 
> 
> Dear @lishunde, @kwagyeman , @rarazac, @JonatanAntoni ;
> 
> I very thank everyone; I am trying to do it as your tutorial. I have a question;
> So, Is there any connection between CMSIS_5 and backends in the Arm? Because, There are many layer (including Convolution, Maxpoolling, etc) in the CMSIS_5. Please, give me an idea or any document about it!
> Thank again!

I am not sure if there is a doc for them, check the source code for both maybe?

If you are interested to run NN in MCU in a similar way to Keras, please check this lib. 
[NNoM](https://github.com/majianjia/nnom)
Which uses CMSIS-NN as backend but providing very easy-to-use APIs similar to Keras. 
Allows you to build state-of-the-art models such as DenseNet, Inception, ResNet very easily.

felix-johnny, 2021-01-07:
@babatn1  Probably it is not relevant now, but if you are still interested, https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro/kernels/cmsis-nn has info on how to get started. 

CMSIS-NN provides optimization for a subset of operators that are supported by TensorFlow Lite Micro. The complete list for TFLu supported operators are in https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/micro/all_ops_resolver.cc .
The list for CMSIS-NN optimized operators can be found in https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN#tfl-micro-compliant-apis 
   Hope that explains your question. I am closing this ticket, but if there is any follow up you would like, please feel free to re-open.

closed 2021-01-07 by felix-johnny