ARM-software/CMSIS_5 #490: Pure C Backend/Core

rsaxvc, 2018-12-27:
Several times now I've wished I could use CMSIS-DSP on something other than a Cortex processor.

Good examples are a desktop simulation of an entire ARM-based product, running unit-tests on a desktop at build-time instead needing to execute them on a micro, and supporting my company's ARM11-based products.

The C backend wouldn't need to be optimized for anything in particular, just implement the same CMSIS-DSP APIs.

A brief survey shows that arm_math.h would need to be updated not to error out when encountering an unknown core, and arm_bitreversal_16/32 would need to be ported to C, but that's about it.

ReinhardKeil, 2019-01-07:
What is exactly that you are looking for?

The Cortex-M0 implementation of the CMSIS-DSP functions are pure C implementations.  Can you not use those in your environment.

rsaxvc, 2019-01-07:
I'd like to be able to use the same CMSIS-DSP APIs on some older ARM cores(Acceleration for ARM11 cores would be appreciated, but out of scope for this request), as well as on Windows(where performance isn't an issue for me, as it would only be used for unit-testing).

Unless I'm reading and building it wrong, the Cortex-M0 implementation of CMSIS-DSP isn't quite C - the FFT uses assembly to do the bitreversals, so I think another define like ARM_MATH_C or ARM_MATH_UNKNOWN_CORE could be used to compile in a 'C' version of the FFT bitreversals and that should handle everything.

christophe0606, 2019-05-24:
@rsaxvc It you want to compile on Windows, Commit 9e28408ac965c528f6771d479abad505833683fd should help.

If you want to compile a pure C version, the bitreversal is now in C. 

If ARM_MATH_DSP is not defined, you should get pure C versions and have no problem building.

rsaxvc, 2019-06-01:
Thanks! This is exactly what I was after.

closed 2019-06-01 by rsaxvc

justiceamoh, 2020-10-12:
Hi, I am trying to do this same thing -- compile a pure C version to simulate CMSIS DSP (and CMSIS NN) on my laptop -- but I'm struggling setting up the build environment. Following the instructions on the DSP page, I created a minimal CMakeList that looks like: 

```cmake
cmake_minimum_required (VERSION 3.14)

set(CMAKE_C_COMPILER "gcc-10")
set(CMAKE_CXX_COMPILER "g++-10")

# Define the path to CMSIS-DSP
set(ROOT /path/to/repo/CMSIS_5)
set(DSP ${ROOT}/CMSIS/DSP)
set(CMAKE_TOOLCHAIN_FILE ${DSP}/Toolchain/GCC.cmake)
set(ARM_CPU "cortex-m0")

# Define the project
project (testcmsisdsp VERSION 0.1)

# Add DSP folder to module path
list(APPEND CMAKE_MODULE_PATH ${DSP})

# Load CMSIS-DSP definitions.
add_subdirectory(${DSP}/Source bin)
```

I run into a few compile errors: `unrecognized command-line option '-mlittle-endian'` and ` unrecognized command-line option '-mthumb'`, which seems to indicate the sources are still expecting an arm compiler, even though the target is set as `cortex-m0`. 

How did you build yours @rsaxvc ? Did you have to isolate certain source files and build outside of the provided build configurations?

rsaxvc, 2020-10-15:
@justiceamoh , I built outside of the provided configurations.

justiceamoh, 2020-10-15:
Thanks @rsaxvc. I was able to compile it as well with help from @christophe0606 via #1015.