ARM-software/CMSIS_5 #491: Open .pb file(Tensor Flow file) using CMSIS-NN

keyur999, 2019-01-03:
Hi there,

How to open tensorflow .pb file using CMSIS-NN?

As I know if i want to open .pb file using arm-NN then i am using CreateNetworkFromBinaryFile Api for the same.

Ex (Using Arm-NN):
armnn::INetworkPtr network = parser->CreateNetworkFromBinaryFile("model/TSR_3Conv3FC_36C_48x1_50ep.pb",{ {"inputs/X", {1, 48, 48, 1}}},{ "FC3/fully_connected/BiasAdd" });

Is there any ways for CMSIS-NN?

Thanks,
Keyur

atrevely, 2019-01-18:
This would be great functionality to have if it doesn't already exist. I'd like to be able to train a network in Tensorflow/Keras and import the output from the freeze_graph function into CMSIS-NN, but I too am having trouble finding a way to do this.

felix-johnny, 2019-06-05:
@atrevely I understand that an interpreter in CMSIS invoking CMSIS-NN APIs would be nice to have, but this is unfortunately not in the scope and doesn't exist. Also as @majianjia mentions(implicitly), asymmetrically quantized tensorflow models need to be converted into their symmetric counterpart before use in CMSIS-NN. 

Here is an example that could be of help. https://github.com/ARM-software/ML-examples/blob/master/cmsisnn-cifar10/code/m7/nn.cpp 

Apologies for the delay in response.

mrgabich, 2020-09-17:
Hy dear

I have a little question.
How to import a new Tensorflow model? For example a Mobilenet?

Best Regards
Geancarlo Abich

majianjia, 2020-09-17:
@mrgabich 
As discussed earlier, I don't think there is a straight forward way, but you can try my [NNoM](https://github.com/majianjia/nnom).

felix-johnny, 2020-09-23:
@mrgabich You can use int8 tensorflow models with CMSIS-NN following some existing examples(none of which is Mobilenet) in https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro/kernels/cmsis-nn . 
Hope that helps.

felix-johnny, 2020-09-23:
@mrgabich  To be a bit more clear, you need some kind of interpreter of the model always with CMSIS-NN and since it is a TFL model that you are interested, making use of the existing integration of TFL micro with CMSIS-NN would be a very good way forward.

closed 2021-01-07 by felix-johnny