ARM-software/CMSIS_5 #492: SVDConv: Generated header does not check defined _ARMCC_VERSION

simtind, 2019-01-03:
From https://github.com/NordicSemiconductor/nrfx/issues/49 : 
In headers generated by SVDConv, the following pre- and postambles are added around the peripheral structure declarations.

```
/* ========================================  Start of section using anonymous unions  ======================================== */
#if defined (__CC_ARM)
  #pragma push
  #pragma anon_unions
#elif defined (__ICCARM__)
  #pragma language=extended
#elif defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050)
  #pragma clang diagnostic push
  #pragma clang diagnostic ignored "-Wc11-extensions"
  #pragma clang diagnostic ignored "-Wreserved-id-macro"
  #pragma clang diagnostic ignored "-Wgnu-anonymous-struct"
  #pragma clang diagnostic ignored "-Wnested-anon-types"
#elif defined (__GNUC__)
  /* anonymous unions are enabled by default */
#elif defined (__TMS470__)
  /* anonymous unions are enabled by default */
#elif defined (__TASKING__)
  #pragma warning 586
#elif defined (__CSMC__)
  /* anonymous unions are enabled by default */
#else
  #warning Not supported compiler type
#endif
```

```
/* =========================================  End of section using anonymous unions  ========================================= */
#if defined (__CC_ARM)
  #pragma pop
#elif defined (__ICCARM__)
  /* leave anonymous unions enabled */
#elif (__ARMCC_VERSION >= 6010050)
  #pragma clang diagnostic pop
#elif defined (__GNUC__)
  /* anonymous unions are enabled by default */
#elif defined (__TMS470__)
  /* anonymous unions are enabled by default */
#elif defined (__TASKING__)
  #pragma warning restore
#elif defined (__CSMC__)
  /* anonymous unions are enabled by default */
#endif
```

In the end section, the code does not check wether the __ARMCC_VERSION version macro is defined yet.

JonatanAntoni, 2019-01-07:
Hi @simtind,

thanks for pointing this out.

Cheers,
Jonatan

JonatanAntoni, 2019-03-12:
Hi @simtind, the issue should have been fixed in the meanwhile. May I ask you to check with the latest version again and finally close this issue, please?

simtind, 2019-03-13:
Looks fixed in SVDConv 3.3.25, thanks!

closed 2019-03-13 by simtind