ARM-software/CMSIS_5 #495: unexpected nan result in matrix multiplication

guglie, 2019-01-07:
I’m using `arm_mat_mult_f32()` to multiply two `arm_matrix_instance_f32` and the result often contains **nan** or **inf** values instead of the expected result.
The function always returns `ARM_MATH_SUCCESS`.

The following are some test values (multiplying a 3x3 matrix for a 3x1 vector):

```
A = [[2 0 0]         
     [0 2 0]   
     [0 0 2]]       

B = [-0.73528117
     -0.77726471
      0.50126243]
```

A*B gives `[nan nan nan]`
instead of `[-1.47056234 -1.55452943  1.00252485]`

---

```
A2 = [[399   1   0]
      [  1   2   0]
      [  0   0   2]]

B2 = [ 4.35113907e-06
      -9.91151571e-01
      -9.98287737e-01]
```

A2*B2 gives `[-16765520183296.0 nan -inf]`
instead of `[-0.98941547 -1.98229879 -1.99657547]`

---

```
A3 = [[2 0 0]
      [0 2 0]
      [0 0 2]]

B3 =  [-0.73528117
       -0.77726471
        0.50126243]
```

A3*B3 gives `[-7.206210753315996e+18 nan -7.679847377213063e+18]`
instead of `[-1.47056234 -1.55452943  1.00252485]`

Is this a sign of some common error in using the library?
I hope I have done some stupid mistake. 
I'm using the _libarm_cortexM4lf_math.a_ precompiled library.

guglie, 2019-01-08:
Hi @JonatanAntoni and @MatthiasHertel80, thanks in advance for your attention! 
I can provide more details if needed!

I've tested with version 4 and version 5.

I'm running on a STM32F4  (cortex-M4)

christophe0606, 2019-05-29:
@guglie We have not been able to reproduce the issue so far. But we will look at it again.
If you have more details they are welcome.

guglie, 2019-05-29:
Thanks @christophe0606, what do you need to reproduce it?

christophe0606, 2019-06-11:
@guglie Do you have a small code we could just build and see how it is behaving ?

closed 2019-12-17 by christophe0606