ARM-software/CMSIS_5 #499: Check asm volatile flagging - mostly excessive, but some may be missing

kjbracey-arm, 2019-01-08:
Scanning the codebase, I can see a load of clearly unnecessary `asm volatile`s - `__REVSH`, `__RBIT`, all `__UXTAB16` et al.

All of these are instructions with no side-effects, so can be treated like normal code, subject to elimination if their results are not accessed or the object they manipulate is not visible - this would be consistent with @JonatanAntoni's comment here: https://github.com/ARM-software/CMSIS_5/issues/463#issuecomment-448926726

Some like `__SSUB16` are justifiably volatile if we want to access the Q or GE flag side-effects, but then there seem to be some missing `volatile` like `__SSAT16`.

`__LDA` and `__STL` arguably have no need for `volatile`, once the missing barrier from #494 is present,  but taking off the `volatile` might put you at the cutting edge of optimisation there: https://stackoverflow.com/questions/45960387/why-dont-compilers-merge-redundant-stdatomic-writes

However, `__LDA` and `__LDRT` are all declared with volatile pointers as parameters, which rather dictates that they must do the access, which mandates a `volatile` on the assembler.

In C++11, there are volatile and non-volatile overloads for `atomic_load`, and CMSIS could do the same for its load/store intrinsics:

__STATIC_FORCEINLINE uint32_t __LDRT(volatile uint32_t *ptr)
{
    uint32_t result;

   __ASM volatile ("ldrt %0, %1" : "=r" (result) : "Q" (*ptr) );
   return(result);
}

#ifdef __cplusplus
__STATIC_FORCEINLINE uint32_t __LDRT(uint32_t *ptr)
{
    uint32_t result;

   __ASM ("ldrt %0, %1" : "=r" (result) : "Q" (*ptr) );
   return(result);
}
#endif

Clearly this is low-priority, as there's nothing actually "wrong" except maybe `SSAT16` - the rest are just maybe suboptimal.

JonatanAntoni, 2019-08-02:
Hi @kjbracey-arm,

I concluded on your PR's and merged them all. Can you please cross check if we hit everything?

Thanks,
Jonatan

kjbracey-arm, 2019-08-12:
Closed all the PRs - I'm not aware of any remaining issues.

Make sure to let me know if any problems occur due to those changes.

closed 2019-08-12 by kjbracey-arm