ARM-software/CMSIS_5 #520: Cortex-M23 / M33 Include Files: FPB and Debug Identification Block registers are not described

yorbeeCMIS, 2019-01-17:
I would like to understand why the core_cm23.h and core_cm33.h CMSIS5 header files do not described the registers of the two following blocks:
* FPB
* Debug Identification Block

We have some DWT registers described but not the above ones.
Can you please clarify why?

JonatanAntoni, 2019-01-18:
Hi @yorbeeCMIS,

we have only the registers defined that are typically used by applications. FPB and Debug Identification registers are mostly used by (external) debuggers but not internally.

Do you have a use case in mind where you would benefit from having this registers defines in CMSIS? If we can clarify about the value of adding them, we can of course do it.

Feel free to provide pull requests for whatever enhancement you think valuable.

Cheers,
Jonatan

yorbeeCMIS, 2019-01-18:
Hi @JonatanAntoni ,

Thanks for the prompt follow up.

I fully agree on the fact this corresponds more to debuggers use.
Then why you did provide DWT and Debug Control Block registers in CMSIS?

JonatanAntoni, 2019-01-18:
Some information from DWT is used for instance by EventRecorder, e.g. as a timing source.

yorbeeCMIS, 2019-01-18:
We even have xPSR registers which are core registers...
OK my point was to understand why some are documented and others are not.

One last question: I think you are missing in cm23 liek you did for cm33 the definition of the Implementation Block Diagram as CM32 has both ICTR and ACTLR registers. We should have an equivalent structure as you have did for CM33

JonatanAntoni, 2019-01-18:
Please consider CMSIS only as a starting point. Its neither complete nor perfect. If you spot shortcomings providing contributions in terms of pull requests is highly appreciated.

Thanks for getting in touch and providing your feedback.

Regarding `ICTR` and `ACTLR` registers it seems you are right. The `core_cm23.h` file misses these. According to the Cortex-M23 TRM only `ACTLR` is available providing only a single bit `EXTEXCLALL`. We can take this as a viable enhancement to the backlog. Please feel free to contribute.

yorbeeCMIS, 2019-01-18:
I think ICTR is also present as it is described in ARMv8-M as ALWAYS IMPLEMENTED even if it is not referred in the CM23 TRM. Could you confirm?

MicroConsult, 2019-02-01:
Regarding the ICTR I had teh same trouble and thus opened a case in ARM developer against the TRM of M23.
I checked the adress 0xE000E004 of the ICTR an a Microchip SAML11 (content=1) and a Renesas S1JA (content=0) and found out, that the Chip design seems to include the ICTR and only the TRM of M23 does not describe its existance.
In the meanwhile you might include the ICTR as part of a SCnSCB structure like it is done in the core include files of the M3/M4/M7.
Br Remo

JonatanAntoni, 2019-02-04:
Hi all,

yes, we need to add the missing registers to `core_cm23.h` (like in `core_cm33.h`) in accordance to Armv8-M ARM. 

Cheers,
Jonatan