ARM-software/CMSIS_5 #522: RTX5 SCVD Changes Removed Objects listing

kevingoh, 2019-01-22:
I was having E306 errors when viewing RTX5 with Component Viewer in Keil MDK 5.26, with the SCVD downloaded with CMSIS pack 5.4..then I tried to add the SCVD file in this repository. 

The error is gone but it seems that it was because the latest commit has omitted the objects listing. 

May I know is this kind of temporary workaround? Because it is now hiding meaningful information...

Attaching screenshot from Vim diff on the thread object portion.

BR,
Seng Tak
[image]

kevingoh, 2019-01-22:
> Hi Seng Tak,
>
> all information is still there. The removed section you are pointing out was used just to read the object name string only (as the comment was suggesting). Object name sting is now printed out using %N token, so this section is not needed anymore.
> What you see in RTX5 component view window should not change since the last (scvd file) version.

Hi Vladimir,

Thanks for replying and sorry for not looking close enough.

But in fact the component viewer was not showing running threads…or was it because I am using the SCVD with RTX kernel v5.4?

VladimirUmek, 2019-01-22:
Yes, this is the reason. The latest RTX5 component view uses different objects IDs, because they were changed recently in RTX5 (see [commit](https://github.com/ARM-software/CMSIS_5/commit/87c33cf147a1c55936401034eb35c33571e053b8).

kevingoh, 2019-01-22:
Thanks a lot Vladimir, sorry for my ignorance

closed 2019-01-22 by JonatanAntoni