ARM-software/CMSIS_5 #525: [Feature request] Support for BNN

dizcza, 2019-02-02:
I'd like to see a binary neural network (BNN) implementation as a good example of CMSIS NN package. 

In BNNs you can pack 8 binary values in a single int8_t value and then use XOR operation that replaces convolution. The authors state that it reduces CPU cycles in x8 times and memory footprint in x32 times.

alessandrocapotondi, 2019-03-04:
Dear @dizcza,

we did it so far for our research purposes, but it is not integrated into the mainstream.

You can have a look here:
https://github.com/EEESlab/CMSIS_NN-INTQ

The library supports also int-8, int-4, int-2, BIN (int-1).

If you are interest to use it or you find some problem on the usage of it you can directly rise issues on the repo.

Best,
Alessandro

alessandrocapotondi, 2019-03-13:
Dear @dizcza, 

if you are interested you just release a mobilenet over CMSIS-NN for STM32H7.
https://github.com/EEESlab/mobilenet_v1_stm32_cmsis_nn

The application rely on an extended CMSIS implemetation which support an asymmetric quantization uint8 (as it done by tensorflow).

You can find the library here:
https://github.com/EEESlab/CMSIS_NN-INTQ/tree/intq-v2.0.1

And a short article here:
https://medium.com/@manuele.rusci/running-mobilenet-on-stm32-mcus-at-the-edge-e217db934f83

dizcza, 2019-03-13:
@alessandrocapotondi 
It looks very interesting! I have STM32F429 board but since you use CubeMX to generate the project (special thanks for this), it should be possible to migrate your solution on other boards as well.

I'll leave this channel open for general discussion and fire issues on your direct git projects if I get any implementation-specific questions.

In the meanwhile, have you tried to compare CMSIS NN solutions (including yours) with https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/experimental/micro? It'd interesting to compare the benefits one solution brings over the other.