ARM-software/CMSIS_5 #528: Incorrect MPU TEX setting for Device non-shared memory

negative-slack, 2019-02-06:
https://github.com/ARM-software/CMSIS_5/blob/80d36a90b1d4dea76ce8652faa187ca36d006653/CMSIS/Core/Include/mpu_armv7.h#L137

That comment (and the code) is incorrect for non-shared device memory. It should be TEX=0b001.

JonatanAntoni, 2019-02-06:
Hi @negative-slack,

You're partly correct, the comment is wrong. It should state:
`* - TEX: 000b (if shareable) or 010b (if non-shareable)`

But the code seems to be correct.
```
#define ARM_MPU_ACCESS_DEVICE(IsShareable) ((IsShareable) ? 
/*                                    TEX  S   C   B                              TEX    C  B  S   */
/*      shareable: */ ARM_MPU_ACCESS_(0U, 1U, 0U, 1U) : // Table 4.55 Column 2  [ 0b000, 0, 1, x ]
/*  non-shareable: */ ARM_MPU_ACCESS_(2U, 0U, 0U, 0U))  // Table 4.55 Column 14 [ 0b010, 0, 0, x ]
```

Cheers,
Jonatan

negative-slack, 2019-02-06:
You're right. The table color code in the link I pasted above is misleading. In the following link it is more clear.
http://infocenter.arm.com/help/topic/com.arm.doc.dui0646a/BIHIIJDC.html#BIHHEDAE

JonatanAntoni, 2019-02-06:
Thanks for double checking this. I'll fix the comment.

JonatanAntoni, 2019-03-11:
@negative-slack, may I ask you to cross check my fix and close the issue, please?

JonatanAntoni, 2019-05-21:
Closing due to no further activity.

closed 2019-05-21 by JonatanAntoni