ARM-software/CMSIS_5 #532: MPU v7 bug in ARM_MPU_ACCESS_NORMAL macro

negative-slack, 2019-02-12:
https://github.com/ARM-software/CMSIS_5/blob/26dc9a88a2588ad9c67dac8af44f6127e22f63ed/CMSIS/Core/Include/mpu_armv7.h#L157

Hi, this macro seems to have a bug in the passing of the C bit value down to the ARM_MPU_ACCESS_ macro. A right shift by one is missing thus loosing the value that is passed.
It should be (((InnerCp)>>1) & 1U). Alternatively it is possible to drop the mask as these values should never be bigger than 3.

negative-slack, 2019-02-13:
FWIW I tested the below patch and eventually managed to get my cache working.
``` C
- #define ARM_MPU_ACCESS_NORMAL(OuterCp, InnerCp, IsShareable) ARM_MPU_ACCESS_((4U | (OuterCp)), IsShareable, ((InnerCp) & 2U), ((InnerCp) & 1U))
+ #define ARM_MPU_ACCESS_NORMAL(OuterCp, InnerCp, IsShareable) ARM_MPU_ACCESS_((4U | (OuterCp)), IsShareable, ((InnerCp) >> 1), ((InnerCp) & 1U))
```

JonatanAntoni, 2019-02-14:
Hi @negative-slack,

thanks for pointing this out. It looks good to me on the first glance. 
Is there any chance to have a unit test case for that? I'd like to enhance the CoreValidation tests if ever possible in a meaningful way.

Cheers,
Jonatan

JonatanAntoni, 2019-07-30:
Hi @negative-slack,

should be fixed now.
Can you please cross check and close?

Thanks,
Jonatan

negative-slack, 2019-08-24:
Looks good.
Closing.

closed 2019-08-24 by negative-slack