ARM-software/CMSIS_5 #534: How quantization was done for arm_nn_examples/cifar10

parai, 2019-02-16:
Hi:
Trying to study how to quantize a model and run on CMSIS NN.

I create a study script here [cifar10.ipynb](https://github.com/autoas/CMSIS_5/blob/develop/CMSIS/NN/Examples/ARM/arm_nn_examples/cifar10/cifar10.ipynb), and find out that the code/m7 generated by [nn_quantizer.py](https://github.com/ARM-software/ML-examples/blob/master/cmsisnn-cifar10/nn_quantizer.py) can't be used as result is bad(the generate q7 bias of convolution layer is all 0), but the default one is good.

So generally I think that the arm_nn_examples/cifar10 are manually generated with special handle of bias_shift and out_shift to achieve better accuracy. But this is a horrible job for new beginners like me.

And another point is that I find that maybe the model [cifar10_m7_iter_300000.caffemodel.h5](https://github.com/ARM-software/ML-examples/blob/master/cmsisnn-cifar10/models/cifar10_m7_iter_300000.caffemodel.h5) is not matched with the one used by arm_nn_examples/cifar10 as the weights is different as checked by my study script  [cifar10.ipynb](https://github.com/autoas/CMSIS_5/blob/develop/CMSIS/NN/Examples/ARM/arm_nn_examples/cifar10/cifar10.ipynb).

So I am thinking can anyone give an example about cifar10 that combined the [ML-examples/cmsisnn-cifar10 ](https://github.com/ARM-software/ML-examples/tree/master/cmsisnn-cifar10) and [ARM/arm_nn_examples/cifar10](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN/Examples/ARM/arm_nn_examples/cifar10) together, and this kind of example is much more helpful and valuable for new beginners.

Thanks.

majianjia, 2019-02-17:
Hi parai,

From my brief view, the problem is on your quantisation function. 
you are doing quantisation to floating which i dont understand quite well. 
~~~python
CONV1_WT = q2f(weights['CONV1_WT'],7)
CONV1_BIAS = q2f(weights['CONV1_BIAS'],7)
compare(net.params['conv1'][0].data.transpose(0,2,3,1), CONV1_WT)
compare(net.params['conv1'][1].data, CONV1_BIAS)
~~~
For example,
In your notebook, the ranges of weights['CONV1_WT'] are in (-0.x, 0.x)
The quantised CONV1_WT should be in range of (-128, 127), but when you do the q2f, which is
~~~python
 D.append(float(v*math.pow(2,-Q)))  # why the -Q instead of Q?
~~~
This function actually returns a much smaller value. 
i think you need a f2q instead of q2f. 

I am not using caffe but I do happy to disscus more 
QQ2759626.

parai, 2019-02-17:
Hi majianjian:
I think you have a misunderstanding of my notebook, the wieghts is the one parsed from the generated code file ML-examples/cmsisnn-cifar10/code/m4/weights.h, so the weithts is in Q Format, so that is I need to convert it back to float and compare it with the float weights of caffe model.
As the picture shows, generally the Q Format weights is matched with the caffe model float weights.

I will add your QQ as friend to chat with you.

majianjia, 2019-02-18:
It was my mistake, you are right. 
I think it is because the weight.h you used to convert back to floating point isn't the one that generated by the model you use.

parai, 2019-02-19:
Hi:
With more research with my notebook [cifar10.ipynb](https://github.com/autoas/CMSIS_5/blob/develop/CMSIS/NN/Examples/ARM/arm_nn_examples/cifar10/cifar10.ipynb)(check at the bottom), I find that for the cifar10 caffe model, the training data feed to the data layer is with range from (float32)-128 to (float32)+128, and for the arm CMSIS NN, as q7 format, so I think the data input can only be in range about from -0.9999... to 0.999..., so by default the cifar10 model can't run on arm CMSIS NN, 
So now I am thinking add a scale layer after the input data layer to scale the input data to range from -1 to 1 and check the results again.

majianjia, 2019-02-19:
Yes, this process is necessary. 
It is also needed to quantize the input data. Same quantization process is needed on MCU when the network is deployed.
It can be simply done by multiplying the input to the power of the input shifting, then clipped to maximum range the -128 to 127.

parai, 2019-02-23:
Hi @majianjia:
By this [mnist notebook](https://github.com/autoas/CMSIS_5/blob/1af1ee0f17c41475198f5537ecee99a781fa0568/CMSIS/NN/Examples/ARM/arm_nn_examples/mnist/mnist.ipynb), I am trying to learn the CMSIS NN things deeply, but I find out that the overflow really has a big affect to the results, as the notebook shows, I think only the first 3 layers result is acceptable, the result of FC layer is really very bad.
Form your NNOM repository, I see you used a fake-clip layer to ensure that from training, all layer's result is in range (-1,1), is there any paper about it, can you share it to me.

parai, 2019-02-23:
find this paper, https://arxiv.org/pdf/1712.05877.pdf, on reading...

majianjia, 2019-02-23:
you are right, overflowing will dramatically affect the results.
The method to avoid it is to adjust the representative range of the 8bit integer.

Each layer's output range must be checked with all the data you have **after the training**. 
Then, adjust the fake_quantisation's Qm.n for each layer.
Then the model must be trained again for the new Qm.n.
Repeat above until none of the output overflowed. 

For example, 
If I have a  3-layer model, and I set them all to Q7.0 at the beginning. The initial training outputs are:
L1 -> -5 to 63 
L2 -> -128 to 240  (this will cause saturation with Q7.0)
L3 -> -23 to 31
The best Q formats for them before overflowing are 
L1: Q6.1 
L2: Q8.-1 
L3: Q5.2
Apparently, the L2 is saturated to Q7.0 which limited the range to -128 to 127. This is where causes the problem. 
Now you should take the new Q format, and put them into your fake_quantisation and train the model again. Some of them might still overflow due to the new setting. Repeat them a few time then you will have the best quantisation for every layer. 

Additionally, from my experience, the training epoch, the batch sizes will affect the output range as well. The larger epoch or the smaller batch will increase the output ranges. 
When every time you adjust these parameters, please check the output range. 

This method can help you to check the output range of every necessary layer with your training or testing dataset. 
[layers_output_ranges(model, x_test)](https://github.com/majianjia/nnom/blob/d6b54305e2926af4c097cce0f84b20e2f4ce94ef/scripts/nnom_utils.py#L246)

According to the fake_clip you mention in my code. 
The quantise range is adjustable, the default range is Q7.0 which is -128 t 127

For the papers, i didn't know much but there are a few issues such as this one are worth to check.
https://github.com/ARM-software/CMSIS_5/issues/327

parai, 2019-02-24:
Hi @majianjia :
Thanks for your help, I can get the mnist trained and then convert the weights and run it on CMSIS NN CPU on host ubuntu, generally it works well, here is the final [notebook](https://github.com/autoas/CMSIS_5/blob/1d97a8a8c0b79a673359aec8910c45714d469bc3/CMSIS/NN/Examples/ARM/arm_nn_examples/mnist/mnist.ipynb).
And I find that the there is big accuracy loss, for the tf model, run on tf, it has accuracy 98% for the test images. But if run on CMSIS NN, there is about 78% of test images can be well predicted with probability above 80%.
I think the loss was resulted by quantization, or the model is not well trained for quantizatin, now I still didn't know well about fake quantization and I don't know how to use it to get the results much more better.

majianjia, 2019-02-25:
it looks like your train/test data is not quantized
the raw data of mnist is 0-255 which should use Q8.-1 format.
The simple least method is mnist/2, to convert it to Q7.0 (0-127), then use both for training and testing
I wil check in detail tomorrow

closed 2019-03-09 by parai