ARM-software/CMSIS_5 #536: Public CI for CMSIS

JonatanAntoni, 2019-02-18:
Hi @JonatanAntoni 
I'm not an expert, but I can help in some way.

First of all I need to understand what are your targets for the library:
- which compilers do you want to support? Which compiler/toolchain do you use now for your tests?
- are you interested in some kind of static code analysis?
- how do you perform unit testing at the moment?

_Originally posted by @escherstair in https://github.com/ARM-software/CMSIS_5/pull/535#issuecomment-464653565_

torfinnberset, 2019-02-18:
For CMSIS-DSP, I have been able to compile and test the library with GCC and clang on both Linux and macOS, as long as we define the macro that we are assuming a Cortex-M0 target. 

This could be a good first step, as it can be done with most CI tools without considering all the challenges of cross-compiling.

For cross-compiled code, I've been working on a Docker container that contains the arm-none-eabi-gcc toolchain and a QEMU emulator for testing ARM binaries, but this is still WIP (and not sure I can release this publically)

escherstair, 2019-02-18:
For the matrix GCC and clang on Linux and macOS (i.e. 4 different build configurations), TravisCI is a good tool.
[https://travis-ci.com/](https://travis-ci.com/)
[A quick tutorial here](https://docs.travis-ci.com/user/tutorial/).

JonatanAntoni, 2019-02-18:
@torfinnberset,

Using GCC for the public part sounds like an option to me. Arm Compiler is a commercial product and hence I would need to check if I can provide the required licenses.

I would try getting started with GCC and QEMU, only. If that works out I can figure out whether we can integrate Arm Compiler and Fast Models which depends on the required licenses.

Cheers,
Jonatan

JonatanAntoni, 2019-02-18:
BTW: Some quick answers to the original questions:

> * which compilers do you want to support? Which compiler/toolchain do you use now for your tests?

Arm Compilers 5 and 6, IAR and GCC are in my personal scope.

> * are you interested in some kind of static code analysis?

Yes, definitively. Internally we apply PC-Lint and evaluated SonarQube and Coverity, all commercial products we won't be able to use on a community basis.

> * how do you perform unit testing at the moment?

I started to add tests for CMSIS-Core a year ago, see https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/CoreValidation. And for CMSIS-DSP there is a test suite available, see https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP/DSP_Lib_TestSuite. Getting them running on a public CI should be possible but would require some work. I have to admit that I won't have much time to invest during the next couple of month.

Please feel encouraged to go ahead. I'll put in as much support as I can give.

Thank you all,
Jonatan

torfinnberset, 2019-02-18:
@JonatanAntoni, both SonarQube and Coverity are available for free for open-source projects. Doesn't CMSIS qualify as this with the Apache 2.0 license?

JonatanAntoni, 2019-02-18:
@torfinnberset, I didn't check if we could get a free license under the discussed circumstances. Please feel free to go ahead if feasible for you. Yes, CMSIS per-se is free and open source under Apache 2.0.

torfinnberset, 2019-02-18:
@JonatanAntoni I just checked on Coverity and SonarQube, and it seems like CMSIS fits the requirements. However, someone from the ARM team needs to link the GitHub project and accept the software agreements.

JonatanAntoni, 2019-02-18:
@torfinnberset, I'll check this with pleasure.

escherstair, 2019-02-18:
Another interesting static code analysis is [codacy](https://opensource.codacy.com/) (which can be used free of charge too for open source projects).

escherstair, 2019-02-18:
After we setup a working TravisCI build, we could integrate [docker containers](https://docs.travis-ci.com/user/docker/) to test cross compilation.

escherstair, 2019-02-20:
If someone is interested I enabled codacy analysis on my fork of CMSIS_5 project.
[Here](https://app.codacy.com/project/escherstair/CMSIS_5/dashboard) you can see the dashboard.
I can see there are not so many serious issues and this means that the quality of the code is high.

Someone from the ARM team can easily enable Codacy analysis on the original github repo, and add a badge to the readme.

SonarQube can give other hints and someone can focus on fix these points.

But I think the most important task is to setup a working CI configuration (either TravisCI or AppVeyor if Windows is an option).

escherstair, 2019-03-08:
@JonatanAntoni news from your side on this topic?

JonatanAntoni, 2019-03-08:
Hi @escherstair,

on the first glance your setup of Codacy looks good and promising. I can take a look what needs me to enable it on the original repo. Thanks for that!

The CI is a more complex situation. I was not able to figure out how to use TravisCI or AppVeyor for CMSIS.

On a quick glance both seems to be suited for native applications but not for cross-compilation. How would one get CMSIS requirements into such a system, i.e. crosscompiler (like gcc-arm or armclang) and target emulators (like qemu or fast models)? 

Next step would be a build system suited for CI. Both seems to rely on tools like CMake for instance. Hence we would need to work out how one could use CMake to build CMSIS projects. 

Can you help with all this?

Cheers
Jonatan

escherstair, 2019-03-08:
I'm not an expert, but I can share my knowledge.
Maybe @torfinnberset can hep too.

ilg-ul, 2019-03-08:
There is no universal solution, Travis and AppVeyor have lots of features and definitely some of them can be used to test CMSIS too.

Travis recently announced test environments for Windows too, so it should be possible to use Travis alone to run tests for all major platforms (Linux/macOS/Windows).

The most convenient platform is probably Linux, it has lots of possible configuration options.

A common scenario is to run a test suite with each pull request, to validate it before being accepted. Actually, once enabled, Travis is triggered for each commit, so nothing skips without being checked.

Simplifying things, all test have a preparation phase, when all prerequisites are satisfied (for example installing a cross toolchain), then a sequence of commands are issued, usually to run some shell scripts which perform the actual tests.

The whole thing can be very complicated, or very simple, depending on how the test is organised.

I extensively used Travis for all kind of tests. An example of a complicated test suite is the one I initially used for my µOS++ framework:

https://github.com/micro-os-plus/eclipse-test-projects

The `.travis.yml` file starts a top script, which starts scripts in each sub-folder, for example:

https://github.com/micro-os-plus/eclipse-test-projects/blob/master/synthetic-posix-tests-micro-os-plus/scripts/travis.sh

The result of an actual run can be seen at:

https://travis-ci.org/micro-os-plus/eclipse-test-projects

---

I agree that the above, although functional, is probably more complicated than it should be.

Thus, for quite some time now, I'm working on an alternate solution to manage projects, called xPack, which also include extensive testability features.

xPacks are actually npm modules customised for C/C++. xPacks can contain source packages, like libraries, but can also contain binary tools, like toolchains, so when dependencies are satisfied, a project is complete and builds can be started.

An example of a project using xPacks for tests is a project template that I wrote for RISC-V devices; the project can generate multiple variants of different projects, for different boards, different syscalls configurations, different trace options, etc, some hundred different choices. 

The test purpose is to instantiate several relevant such projects, and compile them.

The test uses the npm testing infrastructure, and thus it is significantly simpler, the `.travis.yml` does not need to configure anything special, the `npm` defaults are enough.

A run of this test can be seen at:

https://travis-ci.org/micro-os-plus/sifive-templates-xpack

---

For now I have binary xPacks with the ARM Embedded toolchain, the RISC-V Embedded toolchain, Windows Build Tools, OpenOCD, QEMU.

For testability purposes, my µOS++ code is intended to run both on synthetic platforms (like Linux or macOS) and, obviously, on ARM targets, and for this QEMU can be used, the GNU MCU Eclipse QEMU emulates quite a good range of common STM boards.

---

So, for the CMSIS tests, you can first check if the source files compile properly with the ARM Embedded GCC toolchain, preferably with all warnings enabled, and later add run tests using QEMU.

escherstair, 2019-07-30:
@JonatanAntoni 
any news on this direction?

JonatanAntoni, 2019-07-30:
Unfortunately not. Someone needs to volunteer setting this all up. :)
I did similar things in the past and its a lot of work to do which I can hardly afford, personally.

escherstair, 2019-07-30:
I think that ARM-software should take care of this and invest resources in the quality of its product.
Users can voluntary help, but they pay for Keil/ARM software and so they should find high-quality, tested packages.

JonatanAntoni, 2019-07-30:
@escherstair, we do take care. CMSIS itself is free (unpaid) and open source software. There are no plans to invest in publicly accessible CI systems. Nevertheless, I will do my very best to support anybodies initiative.

loverdeg, 2020-08-18:
This project will undoubtedly succumb to whatever risc-v has in store for it if modern development practices are not implemented.

CMSIS enabled Cortex-M to rise to the top. CMSIS is now holding Cortex-M back.

escherstair, 2020-08-25:
@JonatanAntoni one more yeas passed by and no news on this topic.
You wrote

> I will do my very best to support anybodies initiative.

but I think you should do something encouraging this kind of initiative (i.e. promote a contest, ask for volunteers, give publicity to this task ...).
Otherwise I don't think that someone would self-assign this kind of activity

loverdeg, 2020-08-25:
Agree with @escherstair. 

But also know that it is in ARM's best interest to go ahead and fund the effort. CMSIS is far more important than most developers realize.