ARM-software/CMSIS_5 #539: Bug in arm_convolve_HWC_q7_basic.c etc.

nicktao9, 2019-02-21:
In arm_convolve_HWC_q7_basic.c, The line189 define `signed char in_row, in_col;`,as we know,the in_row and col indicate the image location of convolution,just signed char maybe wrong,for example,the row of my image is bigger than 2^7 - 1,the code will run wrong.
The Similar problem in arm_relu_q7.c ,In line96,Defined a `uint16_t i` in order to indicate the number of activating units.but I think the maximum value of i maybe too short。

majianjia, 2019-02-22:
It could be a potential problem.
In one of the very early issues, the widths of those arguments are changed to int16 or uint16. 
The current codes should be working fine with SIMD enabled chips such as M4/M7/M33. 
I have been testing 1D input with one dimension up to 300, and they are working well on my M4 chips. 
However, it seems like they forget to change the loop version for M0/M3

nicktao9, 2019-02-22:
> It could be a potential problem.
> In one of the very early issues, the widths of those arguments are changed to int16 or uint16.
> The current codes should be working fine with SIMD enabled chips such as M4/M7/M33.
> I have been testing 1D input with one dimension up to 300, and they are working well on my M4 chips.
> However, it seems like they forget to change the loop version for M0/M3
> 
> you should be able to make it work simply change them to int16
> 
> ```c
> in_row = stride_y * j + m - padding_y;
> in_col = stride_x * k + n - padding_x;
> ```
> 
> For the relu, i am ok with it. I never met the input larger than 65535 in my case.
> It takes 65k ram which is already larger than my chip's

Thank you for your reply. It's very helpful for me, especially for M4 chips where coordinates do not cross the boundary.

majianjia, 2019-02-22:
I would recommend the [NNoM lib](https://github.com/majianjia/nnom) for a simpler interface instead.

nicktao9, 2019-02-22:
> I would recommend the [NNoM lib](https://github.com/majianjia/nnom) for a simpler interface instead.
amazing！Thank you for your contributions,These things surprise me very much.