ARM-software/CMSIS_5 #540: Decoding error when importing CMSIS_5 using mbed cli

TylerADavis, 2019-02-26:
I am encountering an error when attempting to import CMSIS_5 using the mbed CLI tools. This occurs both when I attempt to add the library to a fresh project (`mbed new` followed by `mbed add`) as well as when I try to import a project that relies on CMSIS_5 (https://github.com/uTensor/uTensor specifically).

I managed to resolve this error in the case of importing uTensor by removing the already imported CMSIS_5 files and then doing `mbed update master` to force the importation of the non development version. I only began encountering this error yesterday while I was able to import the library a few days back, which leads me to believe some recent change caused this break.

I have provided some system info below. If there is anything else I could provide to help with reproducing this, please let me know.

Original error message:
```
(mbed_proj) vagrant@vagrant:/vagrant$ mbed new mbed_proj
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/vagrant" (directory)
[mbed] Creating new program "mbed_proj" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch/tag "latest"
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#ecb3c8c837162c73537bd0f3592c6e2a42994045"
(mbed_proj) vagrant@vagrant:/vagrant$ cd mbed_proj/
(mbed_proj) vagrant@vagrant:/vagrant/mbed_proj$ mbed add https://github.com/ARM-software/CMSIS_5.git
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/vagrant/mbed_proj" (program)
[mbed] Adding library "CMSIS_5" from "https://github.com/ARM-software/CMSIS_5.git" at latest revision in the current branch
[mbed] ERROR: Unknown Error: 'utf-8' codec can't decode byte 0xb5 in position 134: invalid start byte
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
```

Fix:
```
(mbed_proj) vagrant@vagrant:~/my_uTensor/uTensor/external/CMSIS_5$ mbed update master
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/home/vagrant/my_uTensor/uTensor/external/CMSIS_5" (library)
[mbed] Program path "/home/vagrant/my_uTensor"
[mbed] Updating library "CMSIS_5" to branch/tag "master"
[mbed] Updating reference "CMSIS_5" -> "https://github.com/ARM-software/CMSIS_5/#cb3fb9a70f3de3670bf5523a588c7f63cec48ff6"
```

System info:
ubuntu 18.04
```
(mbed_proj) vagrant@vagrant:/vagrant/mbed_proj$ mbed --version
1.9.1
```
```
(mbed_proj) vagrant@vagrant:/vagrant/yolo$ python --version
Python 3.6.7
```
I also reproduced this issue using macOS 10.14.3 using python 3.7.1 and mbed 1.9.1

JonatanAntoni, 2019-02-27:
Hi @TylerADavis,

can you please discuss this issue with [mbed-cli](https://github.com/ARMmbed/mbed-cli), first?
To me it looks like the mbed cli is failing to add CMSIS_5 as a library. We need to figure out why. Later we could introduce changes to CMSIS if we can drill down the cause into this one.

Thanks,
Jonatan

TylerADavis, 2019-03-05:
Seems the underlying issue has been resolved in mbed-cli, and I'm now able to import the latest version of CMSIS_5 to projects, so I'm going to go ahead and close this issue.

https://github.com/ARMmbed/mbed-cli/issues/859#event-2181254120

closed 2019-03-05 by TylerADavis