ARM-software/CMSIS_5 #541: Deploying a 1D Convolutional Network from Keras

atrevely, 2019-02-26:
I've been reading through this repository (and other examples) thoroughly but I'm still unclear on a few particulars related to deploying a neural network with 1D convolutional layers trained in Keras (Tensorflow backend).

1. I assume that 1D convolutional layers can be implemented using the arm_convolve_HWC_q...() functions? For N channels with length M, I assume it'd be identical to a 2D convolution with shape (M, N, 1)

2. I'm still very unclear about how to translate the weight matrix from Keras to one of these layers. If I have data with 2 channels and, say, 4 filters each with kernel size 3, I end up with a weight matrix that's (3, 2, 4). How should this get flattened? In Keras the weight matrix looks as follows:

[[[a1, a2, a3, a4],
 [b1, b2, b3, b4]],

[[c1, c2 , c3,  c4],
 [d1, d2, d3, d4]],

[[e1, e2, e3, e4],
 [f1, f2, f3, f4]]]

Does this turn into:

#define CONV1_WT {a1, a2, a3, a4, b1, b2, b3, b4, c1, ...}?

3. Now there's the quantization part. I assume the whole layer (after flattening) gets quantized together within the same range, but determining the bias_shift and out_shift is killing me, especially for convolutional layers. Is there a straightforward way of deriving the shifts during quantization?

4. Lastly, in Keras I'd use a flatten layer in order to move from the convolution to a dense layer. Does this need to be done by hand?

I realize there's a lot here, but any help with any of these issues would be very helpful for getting my working network over to an M4.

majianjia, 2019-02-28:
From my understanding, 

---

**(1)** You can use those "none-square" 2D for 1D data. simply set H shape to 1 should work. you should use (1, M, N) instead of your what you have said (M, N, 1). 

---

**(2)** Directly flattens the weights will not give you the format for CMSIS-NN, because they are stored in different orders. The weights in keras is (H, W, C, N) in 2D. N is the num of data. So you need this to convert to (N, H, W, C). When you use 1D, weights in keras is (W, C, N), then you need to reorder to (N, W, C) for CMSIS-NN
`cmsisnn_weights = np.transpose(keras_weights, (3, 0, 1, 2))` for 2D
`cmsisnn_weights = np.transpose(keras_weights, (2, 0, 1))`  for 1D

after that, you can store it to file directly, numpy will help you to flatten. 
Check this for more detail [generate_weight()](https://github.com/majianjia/nnom/blob/87bc55e039713f60ac2716443b9c222c8db94cdd/scripts/nnom_utils.py#L146)

---

**(3)** I did confuse about it as well.  

Before you can get the output shift and bias shift, you need to know 4 other formats (Qm.n).
input Qm.n, 
output Qm.n, 
weight Qm.n, 
bias Qm.n

From your weight and bias quantisation, you will have weight Qmn, and bias Qmn. The input and output Qmn will be defined by you manually during the training. The method is presented in [fake_quantisation](https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/Deployment/Quant_guide.md#quantize-activation-data).
I also make my own version [fake_clip](https://github.com/majianjia/nnom/blob/87bc55e039713f60ac2716443b9c222c8db94cdd/scripts/nnom_utils.py#L56) using Lambda layer in keras. 

A nn operation can be define as:
`output = input * weight + bias. `

According to [this issue](https://github.com/ARM-software/CMSIS_5/issues/327), the only important in Qm.n is the n number, which is the fraction part.

We only talk about the exponent, so the * becomeS +.
the relationship between output data's n and shift is :
`output_n + output_shift = (input_n + weight_n)`
So
`output_shift = output_n - (input_n + weight_n)`

the bias is addition, so it should be shift to the same as output format
`bias_shift =  (input_n + weight_n) - bias_n`

to help you understand, `(input_n + weight_n)` is the format in the intermediate calculation, which is q31 before shifting to q7. 

in the CMSIS-NN, output_shift is fixed as right shift. which is actually the inverse num of the above output_shift. so if the above calculation gives you -7, then you should put 7 in the CMSIS-NN. 

All the works above is to make the maximum use of Q7 format without overflowing or saturation
I know it's not easy to understand, but this is how i did it. 

An example using data in the above issues. 
> input Q5.2
> weight Q7.1
> bias Q3.4
> output Q3.4
> 
> input * weight Q28.3
> bias_shift -1
> out_shift -1

According to the equations. 
output_shift = 4 - (2 + 1) = 1 (this will be inverted to put into CMSIS-NN)
bias_shift = (1+2) - 4 = -1

---

**(4)** no, flatten is not needed in MCU, because the data is already flattened in the memory. but you need to know the size of the data. simply multiple all shapes will do. 

Hope it help you

atrevely, 2019-02-28:
I really appreciate the detailed response, it's a big help! I will work through it all and see if I can make headway.

**(1)** I hadn't noticed the "non-square," that looks perfect. Do you happen to know how the layer expects the input? Keras is a little strange and wants inputs paired by timestep, so for example if you have two concurrent measurements x and p, measured from t = 1, 2, ... n, it wants:

[[x1, p1], [x2, p2], ... [xn, pn]]

Does anyone happen to know how I would feed this input into the arm_convolve_HWC_q7_fast_nonsquare() layer?

**(2)** This helps a lot--it looks like if I use `np.transpose(keras_weights, (2, 0, 1)).flatten()` that should put the weights in the right order? They end up being (from the matrix in my original comment):

{a1, b1, c1, d1, e1, f1, a2, b2, ....}

**(3)** I will try and look through this more. I looked over the [Quant_guide.md](https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/Deployment/Quant_guide.md#quantize-activation-data) in the past for doing the quantization but the shifting didn't make sense to me. Especially not for convolutional layers, where it seemed even more unclear how to apply fixed-point arithmetic. Are we still safe applying `
output = input * weight + bias` to convolutional layers?

**(4)** I'm still not totally clear on this. I imagine that I need to do another `np.reshape()` on the output of the pooling layer before feeding it to the fully connected layer?

Thanks again for your help. Hopefully this thread helps others in the future as well.

majianjia, 2019-02-28:
(1) The data you showed is a 1D data. 
It depends on your setting of the batch size. If you feed them in keras, the 1D convolution will take (total batch, batch size, channel) as input shape. In CMSIS-NN, the total batch will always be 1, because each time you run the convolution, it takes only one input. The batch_size here, is the size of you windowed timestamps. If you feeding 1 timestamp each time, you should set it to 1. 
Let say if you put 3 timestamp to a 1D conv, so the batch_size = 3. then your first input to the CMSIS-NN will be: 
[[x1, p1], [x2, p2], [x3, p3]]
The shape in keras is (n/3, 3, 2), correspondingly, the shape in CMSIS-NN will be HWC(1, 3, 2). 
To be noticed, the 1 in (1, 3, 2) is because you are using 2D conv in CMSIS-NN, so the H is constantly 1.

(2) It should be

(3) This part is the most difficult part for me as well. The equation is the same for conv and dense since the weights are quantised all together. 

(4) You can do a reshape, but it will do nothing on MCU.
No matter how is the shape, the data stored in RAM is always flattened or store in order.
Keras will do the same, the flatten() in keras will not move the data, but to change the shape properties from multi-dimension to one dimension.

closed 2019-09-05 by felix-johnny