ARM-software/CMSIS_5 #545: Use of __HeapLimit

KarolisMil, 2019-03-04:
Hello,
Linker script defines __HeapLimit. What should I do with it to have heap properly implemented when using GCC toolchain? Should I write m own _sbrk function for this reason? Why then it is not included in templates. Thank you in advance.

JonatanAntoni, 2019-03-05:
Hi @KarolisMil,

The default startup for GCC on Cortex-M3 can be found [here](https://github.com/ARM-software/CMSIS_5/tree/develop/Device/ARM/ARMCM3/Source/GCC).

The heap is configured in the startup file:
https://github.com/ARM-software/CMSIS_5/blob/6ae6b6689c84001a71dfe6d8f2885c70e4d051d9/Device/ARM/ARMCM3/Source/GCC/startup_ARMCM3.c#L79-L82

The linker script is used to place this static array into device RAM and provides `__HeapLimit` just as a marker:
https://github.com/ARM-software/CMSIS_5/blob/6ae6b6689c84001a71dfe6d8f2885c70e4d051d9/Device/ARM/ARMCM3/Source/GCC/gcc_arm.ld#L165-L171

Please be aware of using `-specs=nano.specs` when linking for a bare metal target.

I hope that helps.

Cheers,
Jonatan

KarolisMil, 2019-03-05:
I understand everything you say very well, but where __HeapLimit is used? Should I implement my own _sbrk? Maybe you can provide any example? I bet default _sbrk implementation does not reference this symbol (__HeapLimit), but only "end". This question might come to my mind due to lack of knowledge of GCC toolchain, but maybe you can enlight me.

JonatanAntoni, 2019-03-05:
@KarolisMil,

Can you please be more specific what you are actually trying to archive? 
What does not work for you with the provided files?

I am not an GCC expert either. As far as I know `__HeapLimit` is not actually used by libc. It only forces a linker error if the defined "stack size" and "heap size" do not fit in RAM. The actual limit between heap and stack depends on your actual usage of both. If this causes trouble for you we might need to dig into the details.

Cheers,
Jonatan

KarolisMil, 2019-03-05:
I am concerned about heap growing too much and overriding stack. As of my understanding of what I have researched, standard _sbrk only checks for heap and stack collision. Image that at the given moment I use 50% of stack. Then I allocate a lot of dynamic memory and it stops allocating only when stack (50% used) is reached. That means it allocated over 100 % of defined heap and goes over stack. Nothing stopping it. Then I use more stack (70%) which corrupts memory which was allocated in heap. This might be solved by having custom _sbrk, which limits allocation depending on __HeapLimit, not on current stack pointer. The question and a proposal is why this is not implemented in

JonatanAntoni, 2019-03-05:
Now I got your point, thanks for clarification. To me it sounds mainly like the question about using GNU libc in bare metal applications.

From CMSIS point of view we don't care about these details too much. So if providing a more elaborated implementation for `_sbrk` solves your concerns there should be nothing stopping you.

If you can come up with a proposal we can discuss whether CMSIS is the right place to share.

JonatanAntoni, 2019-03-05:
cc: @RoboTux can you provide some insights?