ARM-software/CMSIS_5 #547: DSP Lib build projects appear to build only for the M0

ejd037, 2019-03-04:
Hi,

The ARM project for the DSP library, located in both DSP\Projects\ARM\arm_cortexM_math.uvoptx and in DSP_Lib\Source\ARM\arm_cortexM_math.uvoptx, only contains a project for the Cortex-M0 (uVision only shows a "cortexM0l" target). Where are projects for other cores, in particular the M4F? I'd like to make sure the pre-built library that comes with the wad, Lib/ARM/arm_cortexM4b_math.lib, was actually built to make use of the M4F.

Thanks,

Eric

GuentherMartin, 2019-03-05:
Hello Eric,

when you open this project with uVision than you can select different targets (Cortex cores) via the target drop down menu.

Martin

ejd037, 2019-03-05:
Thanks Martin, got it. I'm new to uVision.

JonatanAntoni, 2019-03-11:
@ejd037, can you close this issue if its solved, please?

closed 2019-03-11 by ejd037