ARM-software/CMSIS_5 #548: FLM ProgramPage() API docs need clarification

flit, 2019-03-04:
The documentation for the `ProgramPage()` API of FLM-style flash algorithms needs to explicitly declare whether the API must support programming multiple pages, or if only a single page should be programmed.

Intuitively, it makes sense for the API to program multiple pages, since it has the `sz` parameter. Also, the example code writes as much data as specified by `sz`. However, I've encountered several examples where developers weren't sure, and expected the API to be called once per individual page. Probably at least partially because the API's name is singular, that is, `ProgramPage` versus `ProgramPages`.

I think this all probably comes from the API being originally designed around internal flash technologies where the programming granularity is 1-16 bytes (a phrase). In that case, it's pretty obvious that you wouldn't call the API for each phrase since it's be way too slow. But in the case of modern serial NOR flash (QSPI, Hyperflash, etc), the programming granularity is much larger, on the order of 512 bytes to 2 kB.

JonatanAntoni, 2019-03-05:
Hi @flit,

Thanks for pointing this out. If its feasible for you please provide a proposal of the enhanced documentation. You can find the description of `ProgramPage()` at https://github.com/ARM-software/CMSIS_5/blob/6ae6b6689c84001a71dfe6d8f2885c70e4d051d9/CMSIS/DoxyGen/Pack/src/flash_algorithms.txt#L307-L358

flit, 2019-03-06:
It's on my todo list! 😄

ReinhardKeil, 2019-03-14:
This should clarify ProgramPage https://github.com/ARM-software/CMSIS_5/commit/9f240fd82d3acbc9c4ecc54f2fd8bd1aa8527955#diff-a258ee8a1c58877f6452ba496bd1b732

closed 2019-03-20 by ReinhardKeil