ARM-software/CMSIS_5 #549: Choosing cortex-M3 in iar can compile, but cortex-M4 error

ccnankai, 2019-03-09:
I have a project running on the arm. For convenience, I extracted the code and deleted the files that were not used on the pc, so that I can compile and run directly on the computer with cmake and gcc. After the code is modified, the project is loaded and compiled with iar. When Cortex-M3 is selected, it can be compiled normally. But when you choose M4, there will be a lot of errors, which are inconsistent with the declaration in iar intrinsics.h.

[image]
[image]

closed 2019-03-11 by ccnankai

JonatanAntoni, 2019-03-11:
Hi @ccnankai,

Did you close this issue by purpose?
To me it looks like a clash between files shipped by IAR installation and CMSIS.

Can you give us some more information about the versions you are using?
Perhaps @TTornblom can help as well.

Cheers,
Jonatan

TTornblom, 2019-03-11:
I assume the problem was not including the target specific include file when changing the device in the IDE.
I would be happy to have a look at the project if this is still a problem.