ARM-software/CMSIS_5 #550: v8-M: Contribution proposal for priority boosting functions, PRIS bit in SCB->AIRCR

MicroConsult, 2019-03-13:
I am missing functions to set/clear/get the PRIS bit in SCB->AIRCR.
Use case is the enabling/disabling of priority boosting of armv8-M.
It applies to CM33 and may be added to CM23 as well (after AICRC was added to the CM23.h).

Please find my proposal for an extension of the core_CM33.h header file below.

`#if defined (__ARM_FEATURE_CMSE) && (__ARM_FEATURE_CMSE == 3U)
#define NVIC_SetPriorityBoosting    __NVIC_SetPriorityBoosting
#define NVIC_ClearPriorityBoosting  __NVIC_ClearPriorityBoosting
#define NVIC_GetPriorityBoosting    __NVIC_GetPriorityBoosting

/**
  \brief   Set Priority Boosting
  \details Sets the priority boosting bit using the required unlock sequence.
 */
__STATIC_INLINE void __NVIC_SetPriorityBoosting(void)
{
  uint32_t reg_value;
    
  reg_value  =  SCB->AIRCR;                                     /* read old register configuration    */
  reg_value &= ~SCB_AIRCR_VECTKEY_Msk;                          /* clear bits to change               */
  reg_value  =  (reg_value                                   |
                ((uint32_t)0x5FAUL << SCB_AIRCR_VECTKEY_Pos) |
                SCB_AIRCR_PRIS_Msk                      );      /* Insert write key and priority boosting bit */
  SCB->AIRCR =  reg_value;
}

/**
  \brief   Clear Priority Boosting
  \details Clears the priority boosting bit using the required unlock sequence.
 */
__STATIC_INLINE void __NVIC_ClearPriorityBoosting(void)
{
  uint32_t reg_value;
    
  reg_value  =  SCB->AIRCR;                                     /* read old register configuration    */
  reg_value &= ~(SCB_AIRCR_VECTKEY_Msk | SCB_AIRCR_PRIS_Msk);   /* clear bits to change               */
  reg_value  =  (reg_value                                 |
                ((uint32_t)0x5FAUL << SCB_AIRCR_VECTKEY_Pos));  /* Insert write key                   */
  SCB->AIRCR =  reg_value;
}

/**
  \brief   Get Priority Boosting
  \details Reads the priority boosting bit from the NVIC Interrupt Controller.
  \return  0  Interrupt Priority Boosting is disabled
  \return  1  Interrupt Priority Boosting is enabled
 */
__STATIC_INLINE uint32_t __NVIC_GetPriorityBoosting(void)
{
  return ((uint32_t)((SCB->AIRCR & SCB_AIRCR_PRIS_Msk) >> SCB_AIRCR_PRIS_Pos));
}
#endif`

JonatanAntoni, 2019-03-13:
Thanks @MicroConsult, I'll take a look into that. 
If its feasible for you please consider providing such proposals as a pull request. That would make reviews and integration much easier, i.e. using GitHub review and merge features.

JonatanAntoni, 2019-03-13:
@MicroConsult, the enhancements looks feasible to me.

I am just unsure whether "Priority Boosting" is the best naming. At least we should clearly state that its about prioritizing secure over non-secure interrupts.
Another thing we should clarify is the limitation to __ARM_FEATURE_CMSE == 3U which means compiling with --cmse. Do we expect users running a v8-MML device (i.e. with security extensions) in legacy mode don't need to access this bit?

Please put this change into a pull request if that's feasible for you. If not I try to take a closer look to it next week.

Thanks,
Jonatan

GorgonMeducer, 2019-03-13:
@MicroConsult I like your proposal about set/clear/get APIs. In addition to that, I have a suggestion:

it would be good to replace set/clear/get with set/clear/get/resume. In the new scheme, both set and clear will all have the same return as get, and resume could use that value to resume the previous status.

So, a typical use case would be:

do {
    uint32_t wPreviousStatus = __NVIC_SetPriorityBoosting();
    ….
    __NVIC_ResumePriorityBoosting(wPreviousStatus);
} while(0);

Or

do {
    uint32_t wPreviousStatus = __NVIC_ClearPriorityBoosting();
    ….
    __NVIC_ResumePriorityBoosting(wPreviousStatus);
} while(0);

This kind of structure can also support nesting. 

What do you think?

GorgonMeducer, 2019-03-13:
@MicroConsult By the way, from the actual functionality point of view, rather than "boosting" priority, PRIS is suppressing priorities of NS exceptions.
So, it would be good call it "NSPrioritySuppressing". So, the API would be sound like: EnableNSPrioritySuppressing , DisableNSPrioritySupressing, GetNSPrioritySuppressingStatus and ResumeNSPrioritySuppressingStatus.

What do you think. It is one way to avoid ambiguity.

MicroConsult, 2019-03-14:
- regarding pull request: Will try to fork a repository which seems to be a prerequisite for a pull request

- regarding naming: the phrase priority boosting is not invented by me. It is copied from the arm v8-M TRM document "PRIS, bit [14] Prioritize Secure exceptions. The value of this bit defines whether Secure exception **priority boosting** is enabled." But it is OK for me to choose any other name.

- regarding __ARM_FEATURE_CMSE == 3U: the PRIS bit is only accessable from secure mode and access makes only sense, when there is a NS project in place/or planned thus you need the cmse option anyhow. In a compatibility project without using TZ there is no reason to access the PRIS bit, even if it were available. Hence, stipulating a build error because of missing priority boosting functions might be considered a benefit to pinpoint the unintended usage.

- regarding SW architecture: @GorgonMeducer I like your proposal of set/clear/get/resume very much. The only problem I see is that compatibility to other functions like PriorityGrouping is broken. So the question is, what is considered more important, improved SW architecture or compatibility and thus ease to use. Both were OK for me,

MicroConsult, 2019-03-17:
closed since is dealt with in push request #554

closed 2019-03-17 by MicroConsult