ARM-software/CMSIS_5 #555: Arrangements of weights in CMSIS-NN for all Layerrs

Amardip01, 2019-03-19:
for example,
wt = 
[
    [
        [
            [
                -0.10248074680566788
            ],
            [
                0.03757563233375549
            ],
            [
                0.047999538481235504
            ],
        ]
        [
            [
                0.1954200565814972
            ],
            [
                0.0573272667825222
            ],
            [
                -0.19001096487045288
            ],
         ]
    ]
]

so how the CMSIS weights arrangement shall look like?
I have checked and found that convolution2D and Depthswise convolution API has different arrangements for weights..

wt_cmsis = {1st wt of 1st Kernel, 1st weight of 2nd kernel.....}

or = {1st wt of 1st kernel,2nd kernel of 1st kernel,..etc}

Please confirm.

majianjia, 2019-03-19:
CMSIS-NN uses HWC format. 
so it is defined like this in C code. 
`const q7_t weight[H][W][C] = {...}`

Amardip01, 2019-03-19:
Is it same for Depthwise as well as conv2d?? As per our observation this is not the same arrangement..

majianjia, 2019-03-19:
@Amardip01 
They are the same in CMSIS-NN. 
However, Tensorflow uses CHW to store weights, so you need to convert it to HWC. 

Please search in some previous issues and the code in quantize guide carefully.
Or you can check [here for my implementation](https://github.com/majianjia/nnom/blob/master/scripts/nnom_utils.py#L177)

Amardip01, 2019-03-29:
@majianjia thanks for responding. I have used the portion of your code to generate the weight files.
Issue resolved.

closed 2019-05-13 by felix-johnny