ARM-software/CMSIS_5 #556: Out Shift Calculation in Real Time

Amardip01, 2019-03-19:
Input image will be varied every time in real time. How can we keep OUT_SHIFT constant all the time?
example:
Image 1: output of 1st Layer ranges from [-2 to 1.9873] 
Image 2: output of 1st Layer ranges from [-32 to 31.29] and hence we will take maximum int bits in this case we are taking q5.2 in order to retain maximum number.

Now in real time some 'x' image comes which gives output in the range [-68 to 67.36] by above quantisation this will Generate lot of loss.. 

How to overcome this situation??
Do we also have depthwise separable convolation API for q15?

What is the normal practice? Shall we normalised our every layer output to some fixed numbers; to make sure that values do not overflow..

Please help me address this sitituation..

majianjia, 2019-03-19:
It does not change in real time. Those shifts are defined by your whole dataset. If there is real life image that leads to layer overflows, you should consider to enlarge your training and testing dataset.

If there is an image output are larger than the previous setting occasionally,  it is still ok because the lib uses saturation for these overflow values. 

It is still possible to change it in real time. however, without actually do the conv, you will never know whether it will overflow or not. Simply do it again when it is.

Amardip01, 2019-03-29:
@majianjia understood! Thanks for the responding.

closed 2019-06-05 by felix-johnny